arduino/Arduino #9946: Serial Plotter Plotting Bug

rithviknishad, 2020-03-28:
When plotting using Serial.println and then pausing the Serial.println (just like a one time trigger), and then resetting the arduino to perform it again, the old serial plotted point is not erased and remains static in the plot window.

[image]

rithviknishad, 2020-03-28:
Scnrshot after multiple reset
[image]

What I did to arrive at this error was: (for pausing the SerialPlotter when a SONAR sensor has been activated and it's Echo and Trig Pins are monitored by another arduino)

[image]

nikolradulov, 2020-03-28:
I don't know if this is of any help but the java code for plotter ca be modified such that a clear command can be introduced. In the **~/Arduino/app/src/processing/app/** directory modify the **SerialPlotter.java**  file and add
 `if ( line.equals("<CLEAR>"))
	  {
		  graphs.clear(); // Clear the graphs
		  xCount = 0; // Reset the count of data points
	  }`
in the `public void message` function right after the line trim occurs. In this way you should be able to manually send a clear message and make sure that the graph is cleared even after the resets.
I have not tested this idea on the recent distrib of the IDE,as i just took it from an old project I developed with my friends back in 2017. Hope it helps.

madsdyd, 2020-04-07:
This is not a bug, but by design. The Arduino GUI, which the Serial Plotter is part of, have no way to know if you reset your board, e.g. by pressing a reset button.

@nikolradulov has made a remote command patch in #9953 but this has some overlap with #7461. It seems someone need to decide upon a roadmap or similar for Serial Plotter.