arduino/Arduino #9949: [Library Manager] WiFiNINA_Generic ( Library for WiFiNINA modules/shields to support many more boards )

khoih-prog, 2020-03-28:
https://github.com/khoih-prog/WiFiNINA_Generic

#### New in v1.5.0

1. The original [Arduino WiFiNINA library](http://www.arduino.cc/en/Reference/WiFiNINA) only supports very limited boards, such as: the Arduino MKR WiFi 1010, Arduino MKR VIDOR 4000 and Arduino UNO WiFi Rev.2. This library is based on and modified from [Arduino WiFiNINA library](http://www.arduino.cc/en/Reference/WiFiNINA) to provide support for many more boards, such as SAM DUE, SAMD21, SAMD51, Teensy, AVR Mega, STM32, etc.

## Features

With this library you can instantiate Servers, Clients and send/receive UDP packets through WiFiNINA. The board can connect either to open or encrypted networks (WEP, WPA). The IP address can be assigned statically or through a DHCP. The library can also manage DNS.

jandrassy, 2020-03-29:
it is not a fork, but a copy. the files are renamed and I don't see any useful changes other then K Hoang's name in every file.

khoih-prog, 2020-04-02:
Whoever likes to use will know where to find.

closed 2020-04-02 by khoih-prog

jandrassy, 2020-05-01:
@cmaglie, the library was accepted in Library manager index. is Arduino really OK with this library?