arduino/Arduino #9950: Arduino WiFi Shield

jonjutech, 2020-03-28:
Don't seem to be able to download documentation. The page just shows a page that states it will direct me to arduino.cc. Tried everything I could but couldn't break out of it. I think there is an html issue.

per1234, 2020-03-28:
Please provide the URL of the page where you are having this problem.

jonjutech, 2020-03-28:
https://store.arduino.cc/arduino-wifi-shield

jonjutech, 2020-03-28:
This is where an attempt to download documentation leads me: http://download.arduino.org/products/WIFISHIELD/arduino-wifi-shield-reference-design.zip

per1234, 2020-03-28:
The broken link is the "Eagle Files in .ZIP" button on the "Documentation" tab. The same issue exists for the "Schematics in .PDF" button.

As a workaround, you can still access these files via archive.org's Wayback Machine:
- http://web.archive.org/web/20161027140209/http://download.arduino.org/products/WIFISHIELD/arduino-wifi-shield-reference-design.zip
- http://web.archive.org/web/20170110133955/http://download.arduino.org/products/WIFISHIELD/arduino-wifi-shield-schematic.pdf