arduino/Arduino #9955: [Library Manager] Please add BH1750FVI library

RobTillaart, 2020-03-28:
Please add https://github.com/RobTillaart/BH1750FVI
Thanks!

per1234, 2020-03-28:
Hi @RobTillaart. Each library added to the Library Manager index must have a unique library.properties `name` value (regardless of case). The `name` value used in your library has already been taken:
https://github.com/PeterEmbedded/BH1750FVI

If you would still like your library added to the index, you'll need to change the `name` value to a unique name, then either update the `version` value in library.properties and make a new release/tag or else redo the existing tag.

After that is done, please comment here to let us know we can proceed. Thanks!

RobTillaart, 2020-03-29:
Renamed and updated release tag:    https://github.com/RobTillaart/BH1750FVI_RT    0.1.2

cmaglie, 2020-04-01:
Done.

closed 2020-04-01 by cmaglie