arduino/Arduino #9956: [Library Manager] FlashStorage_SAMD library for SAMD21 and SAMD51

khoih-prog, 2020-03-28:
https://github.com/khoih-prog/FlashStorage_SAMD

### Initial Version v1.0.0
1. Add support to SAMD51 family such as Itsy-Bitsy M4, etc.

Library is based on and modified from:
1. [Cristian Maglie's FlashStorage](https://github.com/cmaglie/FlashStorage)

The FlashStorage library aims to provide a convenient way to store and retrieve user's data using the non-volatile flash memory of microcontrollers.
The flash memory, due to his properties, is generally used to store the firmware code, but it can also be used to store user data.

## Supported hardware

Currently, ***ATSAMD21 and ATSAMD51*** are supported (and consequently every board based on 
this cpu like the ***Arduino Zero, Aduino MKR1000, Nano-33 IoT, Itsy-Bitsy M4, etc***).

cmaglie, 2020-04-01:
Hey @khoih-prog

I've had a Pull Request for the samd51 support here: https://github.com/cmaglie/FlashStorage/pull/27

I didn't merged yet because from the discussion it wasn't clear if this patch was working or not.
Is your fork based on this patch? are you willing to help maintain the FlashStorage library instead of creating a fork?

I see that there is some interest around this library so, if you want, I'm positive to merge the PR above or another alternative impelmentation that is working. I'm wondering also to give write access to some other people to help maintain the library like you, if you wish.

Otherwise I'll just add this library to the list, if you prefer, let me know.

khoih-prog, 2020-04-01:
Your library is a unique and perfect solution in many cases.
It's preferable that you merge any working solution into your library, so that any user can use the new support for SAMD51, (possible more boards in the future, such as some STM32, etc.) and we have only one library.
As I'm working with SAMD51 and in need of the FlashStorage features, I urgently modified your library for my use, then hope the mods will help some people looking for the same feature. This is the final purpose.
I just know about the mentioned patch now. If they are the same, you can use the previous one and ignore this one.

khoih-prog, 2020-04-02:
As the original [Cristian Maglie's FlashStorage](https://github.com/cmaglie/FlashStorage) has just been modified to provide support to SAMD51, I'll close this issue.

closed 2020-04-02 by khoih-prog