arduino/Arduino #9962: Comparison method BUG #9537 - not fixed.

alanmalk, 2020-03-28:
[Screen Shot 2020-03-28.pdf](https://github.com/arduino/Arduino/files/4398002/Screen.Shot.2020-03-28.pdf)
Using a Mac Mini, Mojave 10.14.6.  My first use of IDE 1.8.12

cmaglie, 2020-04-01:
Could you provide a copy of your SKETCHBOOK/libraries folder?

alanmalk, 2020-04-01:
This is part one of two. (Had to split the library folder in two to meet mailing size requirements)--Alan

alanmalk, 2020-04-01:
Can you provide a mechanism to allow me to accomplish this with about 7Mb compressed?Getting this message too many times: "552: 5.3.4 Error: message file too big"
--Alan

per1234, 2020-04-02:
The maximum attachment size allowed by GitHub is 10 MB. Could you put your file on a file hosting service like Dropbox and post a link to it here?

FYI: You can't attach files to a GitHub issue via an attachment to an email reply. GitHub just ignores the email attachment. You need to open the issue in your browser:
https://github.com/arduino/Arduino/issues/9962
and add a reply with your attachment there.

alanmalk, 2020-04-02:
[libraries copy.zip](https://github.com/arduino/Arduino/files/4418548/libraries.copy.zip)

Full library folder attached here as requested.

alanmalk, 2020-04-02:
Full library folder was added to #9962.--Alan

alanmalk, 2020-04-02:
Library folder - compressed - is 7Mb.  I seem to have been able to upload it successfully.
--Alan

cmaglie, 2020-05-13:
@alanmalk thanks for sharing your sketchbook/libraries folder and sorry for the delay.

Unfortunately I'm unable to reproduce the issue, I also tried to unpack your zip on my MacOSX VM sketchbook folder to see if it's something depending on the OS but no errors also there.
I have no clue what the possible issue could be, the comparator now looks correct to me.

Are you still getting the error? I'd like to do more tests if you are available.

alanmalk, 2020-05-13:
Hello Cristian,At this point the error has fallen off my radar.  I doubt that I could reproduce it, and 
actually have no need.
Thank you for investigating but if you are willing to drop the investigation, I would alsobe willing to do the same.
Regards,Alan

cmaglie, 2020-05-15:
>I doubt that I could reproduce it, and actually have no need.

Yes, my fault, I should have acted more promptly...
Anyway, without a way to reproduce it we are basically stuck, so let's close it for now.

closed 2020-05-15 by cmaglie