arduino/Arduino #9969: Possible erros and suggestions

aRbhaSun20, 2020-03-30:
[possible_bugs_in_arduino.docx](https://github.com/arduino/Arduino/files/4402101/possible_bugs_in_arduino.docx)
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.

matthijskooijman, 2020-03-30:
Thanks for your contribution. However, putting your content in an attachment like that is not really ideal, since it makes it harder to participate in this discussion. Also, your issue summary is really generic. Maybe you could put the issue itself in the issue text and update the summary line to be more specific about the issue you are adressing? I had a look at the document and from that it is not immediately clear to me what the issue is exactly, so you might want to also add a bit more detail and maybe an example to clarify.

per1234, 2020-08-04:
Closing due to lack of response.

@aRbhaSun20, if you'll provide the requested information I'll be happy to re-open this issue.

closed 2020-08-04 by per1234