arduino/Arduino #9973: add new deepC library to Library Manager

srohit0, 2020-03-30:
here is the URL of the repository https://github.com/ai-techsystems/arduino.

Thanks,
-Rohit

per1234, 2020-03-31:
Hi @srohit0. I spotted two problems that will prevent the inclusion of this library in the Arduino Library Manager index:

### Library not in root of repository
The library must be in the root of the repository. Your library is currently in the repository's `deepC` folder, which is not allowed.

### Disallowed name
Arduino recently started a new policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

After fixing these issues, update the `version` value in library.properties and create a new release/tag, or else redo the existing release/tag.

Once you have done this, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

srohit0, 2020-03-31:
Thank you @per1234  for the inputs and detailed instruction.

I've fixed the changes. Please check new release at https://github.com/ai-techsystems/arduino/releases/tag/v0.121

thanks,
-Rohit

cmaglie, 2020-04-01:
Done.

closed 2020-04-01 by cmaglie