arduino/Arduino #9977: Pixel Hunt Required to Resize Console Pane

hcgernhardtiii, 2020-03-31:
**IDE version:** 1.8.12
**OS:** Ubuntu Studio 18.04 LTS
**JRE:**  OpenJDK Runtime Environment 11.0.6+10-post-Ubuntu-1ubuntu118.04.1
**DE:**  XFCE 4.12 (GNOME and QT/KDE libs installed)

As the title mentions, resizing the IDE console window requires a pixel hunt.  It would appear that the resize area is a single pixel in height.  

Here's a [Google Drive link](https://drive.google.com/open?id=19xl0Ew-pXI4QNi_4-aiIt-QE_dPn_W4M) to a quick video screenshot of the issue.

As can be seen from the video, there's no real visible resize handle, and what intuitively should be the resize handle is ridiculously small.

Can this be readily fixed?  I'm not a fan of Java, but if the code for this isn't buried to deep and I can get a direction on where to look for this in the source, I can give a shot at finding the issue.

facchinm, 2020-03-31:
Hi @hcgernhardtiii ,
it is due to the desktop integration of Java swing component. Can you test this build https://github.com/arduino/Arduino/pull/9815 ? It uses a brand new look and feel that should solve most of the graphical problems people experience normally.

hcgernhardtiii, 2020-03-31:
Thanks, @facchinm , I just did the install and I have enough pixels of handle to have a visible target.  I'll keep using this build until `aNewReleaseIsPushed || theTestBuildBreaks || iStartWorkingExclusivelyInAssembler`.