arduino/Arduino #9978: Creating a library in arduino manager

DBSStore, 2020-03-31:
I have created a board For education And would like to add the library to arduino manager
github : https://github.com/DBSStore/DBS_AVRIII

per1234, 2020-03-31:
Hi @DBSStore. You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

DBSStore, 2020-03-31:
What do i do next

per1234, 2020-03-31:
I explained what you need to do in my previous reply. I provided links to additional information in case you're not familiar with the process of making a GitHub release.

DBSStore, 2020-03-31:
I created it first, so library.properties has version 1.0.0. Is that correct?

per1234, 2020-03-31:
The choice of version number is up to you, but you need to create a release. Your repository doesn't currently have one:
https://github.com/DBSStore/DBS_AVRIII/releases

DBSStore, 2020-03-31:
What did I do next?

cmaglie, 2020-04-01:
Done, the library is now published.

closed 2020-04-01 by cmaglie