arduino/Arduino #9982: New library: Birdhouse SDK

Nargott, 2020-04-01:
Hi.

I'd like to add my library to Arduino Library Manager. It's available here: https://github.com/Nargott/birdhouse_sdk

Thanks in advance!

per1234, 2020-04-02:
Hi @Nargott. You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Nargott, 2020-04-02:
Thanks for your advice. 
https://github.com/Nargott/birdhouse_sdk/tree/v1.0.0

cmaglie, 2020-04-14:
Done.

closed 2020-04-14 by cmaglie