arduino/Arduino #9987: Cumulocity IoT Library

elpinjo, 2020-04-03:
Hi,

I am busy developing an arduino library for the Cumulocity IoT platform. 
I would like to have this listed in the arduino library manager.
Could you add the library?
The library can be found here: https://github.com/elpinjo/CumulocityClient.git

Kind regards,

Misja

cmaglie, 2020-04-08:
Done.

closed 2020-04-08 by cmaglie

zabutt, 2020-10-26:
I am working on registering a device on cumulocity using ESP32. I am able to connect, register a device and can send data. However, once the device is disconnected, I am unable to reconnect and have to end up creating a new device. I have seen the example in Python where before creating a device, the code checks if the device already exists or not. However, the example in Arduino using different code structure. Can you please help or share some Arduino code that connects and receive device credentials?

ubidefeo, 2020-10-26:
@zabutt 
this is the wrong repo to ask this question.
Please check out the Cumulocity Library repo (https://github.com/elpinjo/CumulocityClient.git) or the Cumulocity website (https://cumulocity.com/guides/concepts/introduction/)