arduino/Arduino #9989: [LibraryManager] Please add this librari

VictorRPacheco, 2020-04-03:
https://github.com/BrinoOficial/BibliotecaBrino

Thanks!

per1234, 2020-04-03:
Hi @VictorRPacheco. In order to be added to the Library Manager index your library must contain a library.properties file. You have a file named [library.properties.txt](https://github.com/BrinoOficial/BibliotecaBrino/blob/master/library.properties.txt), but that won't work. You need to rename the file to library.properties.

After adding the library.properties file, you will need to create a new tag/release or redo the existing one. Make sure that the `version` value matches the tag of the release that contains it.

Once you have done this please comment here so we know we can proceed.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

per1234, 2020-04-06:
Closing as duplicate of https://github.com/arduino/Arduino/issues/10001

closed 2020-04-06 by per1234