arduino/Arduino #9990: Please include this library....

JPF1976, 2020-04-03:
Please can you add this library:

https://github.com/nabontra/ServoTimer2

Thank you.

JP

JPF1976, 2020-04-03:
> Hi @JPF1976 <https://github.com/JPF1976>. There are currently a couple
> issues that will prevent your library from being added to the Library
> Manager index:
>
> In order to be added to the Library Manager index your library must
> contain a library.properties file. The format of this file is documented
> here:
>
> https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format
>
> After adding the library.properties file, you will need to create a
> release <https://help.github.com/articles/creating-releases/> or tag
> <https://git-scm.com/docs/git-tag> that matches the version value in your
> library.properties file.
>
> After doing so, please comment here to let us know we can proceed with
> adding your library to the Library Manager index.
>
> For more information see:
> https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Thank you, I will try and report back to you.

JPF1976, 2020-05-07:
Hi,

Is there a reason the official website has no timer 2 functions?

per1234, 2020-06-14:
Closing due to lack of resolution of the issues I mentioned in my previous reply.

@JPF1976 if you ever get around to making your library compliant, comment here and I'll re-open this issue.

closed 2020-06-14 by per1234

JPF1976, 2020-06-14:
No idea what to do, located another one. Thank you anyway though....