arduino/Arduino #9993: Crashes after moving window to other monitor on macOS

nabelekt, 2020-04-04:
I just updated to 1.8.12 on macOS 10.15.3. I have two external monitors on my MacBook Pro (laptop screen not in use). If I do either of the following, the app immediately crashes:
- Open up a second window (Serial Monitor, Serial Plotter, Library Manager, Preferences, a second sketch, etc.) and move that window to the second monitor, then click back to the first sketch window
- Move the sketch window to the second monitor, then open up a second window (Serial Monitor, Serial Plotter, Library Manager, Preferences, a second sketch, etc.)

Tested in both Dynamic Switching and Discrete Only graphics modes using gfxCardStatus.

per1234, 2020-04-04:
Closing as duplicate of https://github.com/arduino/Arduino/issues/9764

The workaround is to use Arduino IDE 1.8.10, the last version before OpenJDK was introduced. You can download it here:
https://www.arduino.cc/en/Main/OldSoftwareReleases

closed 2020-04-04 by per1234

cmaglie, 2020-04-04:
@per1234 do you think we should put a note in the "new issue" template? or maybe the download page too, at least until it's fixed?

per1234, 2020-04-04:
I think adding a note on the downloads page with a link to the 1.8.10 download is a good idea.

As for adding it to the issues template, we have been getting duplicates, but not so many that it takes a significant effort to close them. I suspect that the number of issues will drop once people are directed on the downloads page to use Arduino IDE 1.8.10. That said, I'm not necessarily against updating the issue template.

ar13kn, 2020-04-08:
Im also facing the same issue, Mines a dual monitor Hackintosh. running Catalina 10.15.4.  Arduino 1.18.12 Crashes on both monitor, especially on secondary monitor mostly when arduino window is moved to the second display. The error is : Crashed Thread: 35 Java: Java2D Queue Flusher.

dirkx-reno-2020, 2020-04-30:
Another work around is selecting the option 'Displays have separate Spaces' in System Preferences -> Mission Control.