arduino/Arduino #9994: Auto download required libraries

sysdl132, 2020-04-04:
<h1>Feature request</h1>
<p>Everytime I must download the libraries manully.So I want request a new feature.</p>
<h2>Auto download the required libraries</h2>

per1234, 2020-04-04:
Am I correct in assuming that by "required libraries" you mean library dependencies of a sketch?

Perhaps the sketch author could define this in sketch.json. That file is not currently used by the Arduino IDE (only Arduino CLI and Arduino Web Editor), but it makes sense to use the established sketch metadata system. Arduino Web Editor already adds an `included_libs` name to that file, which is used to define the versions of the sketch's library dependencies:
```json
  "included_libs": [
    {
      "name": "ArduinoBLE",
      "version": "0.1.2"
    },
    {
      "name": "RTClib",
      "version": "1.3.3"
    }
  ]
```

dsyleixa, 2020-04-05:
I'm afraid I have to disagree about auto-download, because there perhaps might exist several libs of identical libname (libname.h) which might mess up the entire program functionality.

per1234, 2020-04-05:
>there might exist several libs of identical libname (libname.h) which might mess up the entire program functionality.

If the idea is to detect dependencies based on header file names, this is certainly a valid concern. It's likely that approach would do more harm than good.

However, if the feature worked from a list of library names specified by the sketch author, this is not a problem, since every library in the Library Manager index is required to have a unique name. We already have this feature for specifying dependencies of a library via the [library.properties `depends` field](https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format).

sysdl132, 2020-06-30:
OK.thanks for your answer:)

closed 2020-06-30 by sysdl132