arduino/Arduino #10010: [Library Manager] Please add Framebuffer_GFX and dependent libraries

marcmerlin, 2020-04-07:
https://github.com/marcmerlin/Framebuffer_GFX
https://github.com/marcmerlin/FastLED_NeoMatrix/
https://github.com/marcmerlin/SmartMatrix_GFX
https://github.com/marcmerlin/FastLED_SPITFT_GFX
https://github.com/marcmerlin/FastLED_TFTWrapper_GFX
https://github.com/marcmerlin/FastLED_RPIRGBPanel_GFX

per1234, 2020-04-07:
In all your repositories, you need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

The only repository that does already have tags is https://github.com/marcmerlin/SmartMatrix_GFX, but the `name` value in library.properties was still set to the non-unique value "Adafruit NeoMatrix" at the time of the last tag:
https://github.com/marcmerlin/SmartMatrix_GFX/blob/v1.1.2/library.properties#L1
I see that since then you have changed to a unique name, but you haven't created a new tag/release since that time.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

marcmerlin, 2020-04-07:
@per1234  sorry for my being a bonehead. I did tag all the releases on my side, but didn't realize that git push didn't push the tag.
I ran a "git push origin 1.0" in each repo, and they should be good now.

cmaglie, 2020-04-08:
Hi @marcmerlin, I've added your libraries to the indexer, but the following three needs some changes:

 https://github.com/marcmerlin/FastLED_NeoMatrix/
 https://github.com/marcmerlin/SmartMatrix_GFX
 https://github.com/marcmerlin/FastLED_SPITFT_GFX

```
JOB 2888 - 2020/04/08 17:45:20 Scraping https://github.com/marcmerlin/FastLED_NeoMatrix.git
JOB 2888 - 2020/04/08 17:45:22 Checking out tag: 1.0
JOB 2888 - 2020/04/08 17:45:23 path=github.com/marcmerlin, archivename=FastLED_NeoMatrix-1.0.0
JOB 2888 - 2020/04/08 17:45:23 Error syncing library: Error while zipping library: Symlink not allowed: /home/jenkins/jenkins/jobs/Libraries/workspace/gitclones/github.com/marcmerlin/FastLED_NeoMatrix/examples/Aurora -> GFX/Aurora
```

```
JOB 2889 - 2020/04/08 17:45:21 Scraping https://github.com/marcmerlin/SmartMatrix_GFX.git
JOB 2889 - 2020/04/08 17:45:23 Checking out tag: 1.0
JOB 2889 - 2020/04/08 17:45:23 path=github.com/marcmerlin, archivename=SmartMatrix_GFX-1.0.0
JOB 2889 - 2020/04/08 17:45:23 Error syncing library: Error while zipping library: Symlink not allowed: /home/jenkins/jenkins/jobs/Libraries/workspace/gitclones/github.com/marcmerlin/SmartMatrix_GFX/examples/FastLED -> FastLED_NeoMatrix_SmartMatrix_LEDMatrix_GFX_Demos/FastLED
```

```
JOB 2890 - 2020/04/08 17:45:22 Scraping https://github.com/marcmerlin/FastLED_SPITFT_GFX.git
JOB 2890 - 2020/04/08 17:45:25 Checking out tag: 1.0
JOB 2890 - 2020/04/08 17:45:25 path=github.com/marcmerlin, archivename=FastLED_SPITFT_GFX-1.0.0
JOB 2890 - 2020/04/08 17:45:25 Error syncing library: Error while zipping library: Symlink not allowed: /home/jenkins/jenkins/jobs/Libraries/workspace/gitclones/github.com/marcmerlin/FastLED_SPITFT_GFX/examples/FastLED -> FastLED_NeoMatrix_SmartMatrix_LEDMatrix_GFX_Demos/FastLED
```

The issue is that we do not support symlinks inside a library. To fix this for each library:

1) please remove symlinks by just making a copy of the examples
2) commit & push
3) delete and redo the release tag on the new master.

Once done, the libraries should be automatically published within an hour by our indexer.
I'll go ahead and close this issue. If you need further assistance just comment here.

closed 2020-04-08 by cmaglie

marcmerlin, 2020-04-08:
@cmaglie thanks for the info. So, " just making a copy of the examples" isn't really practical because they are big and stored in a git submodule which is not copied multiple times in those libraries that are able to share the same common code.
I've removed the symlinks and replaced them with files that tell you what cd command to type. It's a bit sad, but not the end of the world.

That fixed
https://github.com/marcmerlin/FastLED_NeoMatrix

but those two still seem unpublished. Are you able to see what errors you get on your side?
- https://github.com/marcmerlin/SmartMatrix_GFX
- https://github.com/marcmerlin/FastLED_SPITFT_GFX

Thanks, Marc

per1234, 2020-04-09:
>but those two still seem unpublished. Are you able to see what errors you get on your side?
>- https://github.com/marcmerlin/SmartMatrix_GFX
>- https://github.com/marcmerlin/FastLED_SPITFT_GFX

SmartMatrix GFX has been published:
[image]

FastLED_SPITFT_GFX hasn't been published because you haven't made a release/tag since you removed the symlinks:
https://github.com/marcmerlin/FastLED_SPITFT_GFX/compare/1.1...master
I notice both your 1.0 and 1.1 tags point to the same commit.

If you create another release/tag for the tip of the master branch I'm sure FastLED_SPITFT_GFX will get published on the next run of the indexer job.

>stored in a git submodule

FYI, submodules are not included in the Library Manager installation. So you just end up with an empty `examples\FastLED_NeoMatrix_SmartMatrix_LEDMatrix_GFX_Demos` folder where you would expect the submodule to be. The installation is also not a Git repository, so the instructions in the `examples/README.md`:
>FastLED_NeoMatrix_SmartMatrix_LEDMatrix_GFX_Demos is populated from https://github.com/marcmerlin/FastLED_NeoMatrix_SmartMatrix_LEDMatrix_GFX_Demos/
>
>Please run git submodule update --init --recursive

Won't work for users who did a Library Manager installation (nor for those who used the GitHub generated .zip file).

One solution would be to use a [Git subtree](https://www.atlassian.com/git/tutorials/git-subtree) instead of a submodule.

marcmerlin, 2020-04-09:
@per1234 I see SmartMatrix_GFX too now. Maybe my client cached old indexes, sorry about that.
As for FastLED_SPITFT_GFX , thanks for the hint. I did release 1.1 a few hours ago, but the tag didn't get applied to master, I'm not quite sure why.
I've just released 1.2 and made sure that this time the tag is applied to master, so all should be good. Again, sorry for taking your time with my git ineptitude.

As for submodules, yes, I know they are not going to work with the library manager. That's not a problem, the submodule is not needed in any way, it's simply a lot of example code that's common to 5 different libraries, which is why I factored it out on its own git repo.

git subtree: thanks for the link. It seems like an interesting alternative, but I think it effectively duplicates the data, and each time I update one location, I'd be responsible for updating the subtree in all 5 checkouts that use it and put this 5 times to github (vs once with submodules). I'm not sure that's a win in my case.
However your point is quite valid that I should update my README.md to give an alternative to submodule for people who did a library download, thanks for pointing that out.

and again thanks for your help, and the service you provide, really appreciated.