arduino/Arduino #10017: [Library Manager] Please add LIN Master library

gicking, 2020-04-09:
Please add https://github.com/gicking/LIN_master_Arduino

It provides a LIN master node emulation with preemptive background operation

Thanks!

per1234, 2020-04-09:
Hi @gicking. You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

gicking, 2020-04-12:
hi,

sorry, my mistake! I have now created a release v1.0 and changed library.properties accordingly (previously 0.1). For your support thanks a lot!

Georg

cmaglie, 2020-04-14:
Done.

closed 2020-04-14 by cmaglie