arduino/Arduino #10018: 1.8.12 IDE: java.io.IOException: Cannot run program "C:\Program Files\Arduino\arduino-builder"

VincentPrunier, 2020-04-09:
Hi,
I've just installed the newest ide (title) to reboot old projects and activities (last uses with 1.0.3 a while ago)... So firstly installed:
-On a NEC laptop, vista sp2 business 32, jdk8u241, Uno board (but isn't a cause): installation, board driver, communication with board all OK...
-BUT as soon as I try any program verifying, messages below appear (pointing the first line in program, which is... a comment! Anyway, any programs do the same "before any real program verifying"...)
--------------Messages start-----------------------------

C:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\Arduino\hardware -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files\Arduino\libraries -libraries \\WDMYCLOUD\Public\9_Softwares\Arduino\VP_apps\libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0043 -ide-version=10812 -build-path C:\Users\Vincent\AppData\Local\Temp\arduino_build_635784 -warnings=default -build-cache C:\Users\Vincent\AppData\Local\Temp\arduino_cache_477336 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Program Files\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files\Arduino\hardware\tools\avr -verbose D:\ARCHOS\2016-04\Donnees\Arduino\VP_apps\VPrecupTpStiBastien\VPrecupTpStiBastien.ino

processing.app.debug.RunnerException
	at cc.arduino.Compiler.callArduinoBuilder(Compiler.java:317)
	at cc.arduino.Compiler.loadPreferences(Compiler.java:216)
	at cc.arduino.Compiler.build(Compiler.java:175)
	at processing.app.SketchController.build(SketchController.java:664)
	at processing.app.Editor$BuildHandler.run(Editor.java:1624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: Cannot run program "C:\Program Files\Arduino\arduino-builder": CreateProcess error=193, %1 n'est pas une application Win32 valide
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.Compiler.callArduinoBuilder(Compiler.java:297)
	... 5 more
Caused by: java.io.IOException: CreateProcess error=193, %1 n'est pas une application Win32 valide
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 7 more
... 7 more
--------------Messages end-----------------------------

-Tried to "run" straight that "arduino-builder.exe" just to see: same win kind of message "... isn't a valid win32 app" (OK, I suppose it isn't intended to be run like that... Just to see...)
-Tried to uninstall JDK to have only JRE, uninstall/reinstall IDE... The same!

-Then I tried on my XPsp3 pro 32 desktop, JRE8u71... All the same!

-Then I've seen in #7598 issue a few possible tracks like trying to disable antivirus soft (Avast free in my case!): I've tried that but nothing works!

I'm just desperate... Missed anything? I've seen nothing in forum about that (sorry if I'm wrong...).

Thanking you forwardly for helping me on that!
Regards,
Vincent

VincentPrunier, 2020-04-09:
Me again!

after having tested online ide (neither working due to lack of recognition of the board...), I uninstalled once more 1.8.12, installed 1.8.11 and it's working fine... Enough for the moment, let's play...

VincentPrunier, 2020-04-09:
"closed" to me for the moment, but maybe useful for you...
Take care!

per1234, 2020-04-11:
I believe the issue of 1.8.12 not working, while 1.8.11 does, is specific to Windows Vista. We have had a couple posts about this on the Arduino Forum, including this:
https://forum.arduino.cc/index.php?topic=675024

According to my notes, XP support was lost in 1.8.10, but with a different error message in IDE versions 1.8.10 and 1.8.11:
```
panic: Failed to find GetFileInformationByHandleEx procedure in kernel32.dll: The specified procedure could not be found.
```
So there may now be two issues compounded breaking XP support.

XP-support related forum topics:
- https://forum.arduino.cc/index.php?topic=657164
- https://forum.arduino.cc/index.php?topic=639058
- https://forum.arduino.cc/index.php?topic=656945
- https://forum.arduino.cc/index.php?topic=656079
- https://forum.arduino.cc/index.php?topic=649086
- https://forum.arduino.cc/index.php?topic=639747
- https://forum.arduino.cc/index.php?topic=603700.msg4306138#msg4306138
- https://forum.arduino.cc/index.php?topic=603700.msg4310501#msg4310501
- https://forum.arduino.cc/index.php?topic=603700.msg4310941#msg4310941
- https://forum.arduino.cc/index.php?topic=603700.msg4320929#msg4320929
- https://forum.arduino.cc/index.php?topic=648757
- https://forum.arduino.cc/index.php?topic=653940
- https://forum.arduino.cc/index.php?topic=646011
- https://forum.arduino.cc/index.php?topic=641170
- https://forum.arduino.cc/index.php?topic=655775
- https://forum.arduino.cc/index.php?topic=443558.msg4320985#msg4320985

And about the new error message on XP:
- https://forum.arduino.cc/index.php?topic=667222

I don't have a Windows XP or Vista machine, so I haven't been able to verify any of this.

---
It would be good to decide whether support for Windows XP and Vista has officially been dropped, or whether the intent is to eventually fix these issues. If the former, the statement on the downloads page needs to be updated:
https://www.arduino.cc/en/Main/Software
>Windows Installer, for Windows XP and up

cmaglie, 2020-04-15:
Supporting such older OS becomes harder and harder as libraries, tools, compilers etc. drops support for them. I think we must remove the XP and Vista "supported" claim from the download page. What's the oldest version of Windows that we still support?

cmaglie, 2020-04-15:
I guess the new minimum required version is Windows 7 https://it.wikipedia.org/wiki/Versioni_di_Microsoft_Windows

VincentPrunier, 2020-04-16:
Hello everybody, and thank you all for answering!

Yes, I guess it would be suitable to note :
-a win7 mini. requirement on 1.8.12 (through my modest and unique test on vista32sp1pro: 1.8.11 is OK!)
-and I've just tested my 1.8.11 install on XPproSp3: not even a running IDE! (see attached pic)
[image]

Thank you very much once more, I think modifying the "XP and up" mention should be a nice thing, yes. I close...

per1234, 2020-04-16:
I think we should leave this open, since it acts as our issue to track the need to correct the documentation about which Windows versions are supported.

We will close it once the misinformation on the website has been fixed.

closed 2020-05-08 by cmaglie