arduino/Arduino #10028: [library manager] Please add DS18B20 library

RobTillaart, 2020-04-11:
Please add https://github.com/RobTillaart/DS18B20
Thanks!

per1234, 2020-04-11:
Hi @RobTillaart. Each library added to the Library Manager index must have a unique library.properties `name` value (regardless of case). The `name` value used in your library has already been taken:
https://github.com/matmunk/DS18B20

If you would still like your library added to the index, you'll need to change the `name` value to a unique name, then either update the `version` value in library.properties and make a new release/tag or else redo the existing tag.

After that is done, please comment here to let us know we can proceed. Thanks!

RobTillaart, 2020-04-11:
renamed to https://github.com/RobTillaart/DS18B20_RT

per1234, 2020-04-11:
The issue is about the `name` value in library.properties, not the repository name (there are no restrictions on repository names), though it does make sense to match the two.

The `name` value in your repository is still set to the non-unique value DS18B20:
https://github.com/RobTillaart/DS18B20_RT/blob/master/library.properties#L1

RobTillaart, 2020-04-11:
oeps, fixed

per1234, 2020-04-11:
Now you just need to make a new tag/release or redo the existing one and it should be ready to be added to the Library Manager index.

RobTillaart, 2020-04-11:
It is not my day apparently, sorry, release should be fixed now

cmaglie, 2020-04-14:
Done.

closed 2020-04-14 by cmaglie