arduino/Arduino #10030: [Library Manager] Please add  Arduino FOC library

askuric, 2020-04-11:
https://github.com/askuric/Arduino-FOC

Could you please add the Arduino FOC library to the library manager. 
Thanks a lot,
Antun

per1234, 2020-04-11:
Hi @askuric. Arduino recently started a new policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format

Then update the `version` value in library.properties and create a new release/tag, or else redo the existing release/tag.

Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

Once you have done this, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

askuric, 2020-04-11:
Thanks for the heads up. I had no idea. 
I have renamed the library (Simple FOC) and refactored the code. 
I hope it is in line with the specification. 

https://github.com/askuric/Arduino-FOC

Thanks,
Antun

cmaglie, 2020-04-14:
Done.

closed 2020-04-14 by cmaglie