arduino/Arduino #10032: uploadUsingProgrammer problems should fail gracefully

tedder, 2020-04-11:
I sat and looked at this message for a long time, and upgraded my Arduino version, before determing what specific user error I was making- I had the wrong programmer selected. Based on the issues and other Googling I'm not the first person to get this NPE. It would be helpful if there was a better error message in the console, at a minimum.

I assume that `targetPlatform` is null [at this point in the code](https://github.com/arduino/Arduino/blob/ef8d669f6a49a438430aae3e1f29161e34ce5d63/arduino-core/src/cc/arduino/packages/uploaders/SerialUploader.java#L295), so `targetPlatform.getProgrammer(..);` fails.

```
java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingProgrammer(SerialUploader.java:295)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:90)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2047)
	at java.lang.Thread.run(Thread.java:748)
```

somewhat-related issues: #7412 #7034

per1234, 2020-04-14:
Which programmer did you have selected @tedder?

Which board did you have selected?

tedder, 2020-04-14:
I just spent a **lot** of time trying to repro this, and I was afraid it was going to be that bane of reported bugs that can't be reproduced. But I got it.

Roughly, I have board support for STM32/ESP32/ESP8266/attiny installed, and have a Pololu AVR programmer connected. I couldn't repro it until I used the "ST-Link (STM32L0)" while on an ATTiny board. I then reproed it with the "STM32 BOOTLOADER (STM32L0)". I then changed my board from an ATTiny back to Arduino Uno and still get the NPE.

So it's clearly something provoked by the STM32 programmer.