arduino/Arduino #10038: Ethernet Shield 2 with missing material on Schematics

unusual1, 2020-04-12:
I've recently recieved a Arduino Ethernet Shield 2(stamped with 14-19 time log) but the recieved board doesn't match the presented schematic on the product page: https://store.arduino.cc/arduino-ethernet-shield-2
 For instance, There are some components after the MISO connection pin on the ICSP connector on the board.
In my opinion it could be a new revisison of the Board?
Please upload the last schematic on website...

per1234, 2020-04-12:
Thanks for reporting this issue @unusual1.

I see we have two previous reports that look similar:
- https://github.com/arduino/Arduino/issues/9362
- https://github.com/arduino/Arduino/issues/8395

Please check to see if the issue you are reporting is the same as either, or both of those. If so, please let me know and I'll close this issue so that we can consolidate all the discussion about this problem in a single place.

unusual1, 2020-04-12:
Some other components are missing in my post but the main problem is same. Schematic is not actual.

Android için Outlook<https://aka.ms/ghei36>'u edinin

  *   #9362<https://github.com/arduino/Arduino/issues/9362>
  *   #8395<https://github.com/arduino/Arduino/issues/8395>

per1234, 2020-04-13:
OK, thanks for checking on that for me. I'm going to close this as a duplicate. 

If you have additional information to add to what is in the previous issues, feel free to comment there.

closed 2020-04-13 by per1234

unusual1, 2020-04-13:
Who upload the new schematic? and When?

Android için Outlook<https://aka.ms/ghei36>'u edinin

Closed #10038<https://github.com/arduino/Arduino/issues/10038>.