arduino/Arduino #10041: please remove my CStringBuilder library from the Library Manager index

jandrassy, 2020-04-13:
I removed the release tags 3 weeks ago. Last week I deleted the repository, but the library is still in index.

I moved the CStringBuilder class to my StreamLib library two years ago and there it has fixes for the new cores.

Thank you

cmaglie, 2020-04-14:
Lib removed.

closed 2020-04-14 by cmaglie

per1234, 2020-04-15:
@cmaglie this report:
>I removed the release tags 3 weeks ago. ... but the library is still in index.

seems to contradict what the Library Manager FAQ says:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#sorry-i-did-something-wrong-how-can-i-change-or-unpublish-an-already-published-library
>In order to unpublish a library version, delete its related tag/release.
>
>Modified or Deleted library versions are processed every Sunday.

Is this a bug in the indexer code that needs to be fixed, or does the documentation need to be updated?

cmaglie, 2020-04-15:
@per1234 you're right, due to an internal change to our CDN at the moment to change/remove a previously released library is required an admin. May you change the documentation until we fix it?

per1234, 2020-04-15:
OK, I have now done so:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ/_compare/2fa4c531f25826297186cd59a9ce6c6da23b0a08...cd96c32966955e3f7bfe11110b2eabf83dedfb39

We should be able to just revert that edit once the previous behavior is restored.