arduino/Arduino #10043: [new library] Please register new library https://github.com/Yurik72/ESPHap

Yurik72, 2020-04-13:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.

per1234, 2020-04-13:
Hi @Yurik72. You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Yurik72, 2020-04-13:
Hi per1234, I did that , please check and proceed, otherwise let me know what is wrong, i'm doing this first time, thanks

cmaglie, 2020-04-14:
Done.

closed 2020-04-14 by cmaglie