arduino/Arduino #10047: Mac OS X before 10.10 - dynamic linking broken

bonhomat, 2020-04-15:
#### dynamic linking doesn't work, some functions missing in older Mac OS X versions
since IDE 1.8.10  on Mac OS 10.9.5 there are missing functions the IDE is calling
also in the latest Beta 1.9.0 and Latest Nightly from 25 March 2020 15:34:57 GMT

Probably this is also true for Mac OS X 10.8 systems.

My test system is Mac OS X 10.9.5. With 
```
java version "1.8.0_202"
Java(TM) SE Runtime Environment (build 1.8.0_202-b08)
Java HotSpot(TM) 64-Bit Server VM (build 25.202-b08, mixed mode)
```
The only working version is Arduino 1.8.9.
In my oppinion the info on the page that it works with Mac OS X 10.8 is at the moment not true.

By tring to "Verify" a program there are the follwing issues
error with Arduino 1.8.10 & 1.8.11:
```
dyld: Symbol not found: _unlinkat
  Referenced from: /Applications/Arduino 1.8.10.app/Contents/Java/arduino-builder
  Expected in: flat namespace

/Applications/Arduino 1.8.10.app/Contents/Java/arduino-builder returned 133
```

error with Arduino 1.8.12,  Beta 1.9.0,  Nightly 25 March 2020 15:34:57 GMT:
```
dyld: Symbol not found: _fdopendir$INODE64
  Referenced from: /Applications/Arduino 1.8.12.app/Contents/Java/arduino-builder
  Expected in: flat namespace

/Applications/Arduino 1.8.12.app/Contents/Java/arduino-builder returned 133
```

maybe the headers in this project can help to fix it
https://github.com/macports/macports-legacy-support
everything else works so far fine

facchinm, 2020-04-15:
Related with https://github.com/arduino/Arduino/issues/10018 .
We updated golang for security reason at 1.8.9 time, and this is the reason why the builder stopped working https://golang.org/doc/go1.11#ports . 
@cmaglie should we amend the download page or try restoring the functionality?

cmaglie, 2020-04-15:
I don't think we can easily restore the functionality for macosx<10.10 if Golang dropped support for them, IMHO we should update the download page.

cmaglie, 2020-04-21:
The download page has been updated.

closed 2020-04-21 by cmaglie

zoranmikic, 2020-06-22:
I recently came across this error while installing the Arduino IDE on Mac OS X 10.9.5.  The web page has not been updated, as you say above.  Please look at:

https://www.arduino.cc/en/Main/OldSoftwareReleases#previous

where it says that version 1.8.12 supports "10.8 Mountain Lion or newer", which it does not.  The latest version that runs on my Mac OS X 10.9.5 system is 1.8.9.  The newer ones (1.8.10, 1.8.11, 1.8.12, and 1.8.13) have errors, including

dyld: Symbol not found: _unlinkat

or

dyld: Symbol not found: _fdopendir$INODE64

as you have noted above.  Could you please update that web page, since it causes a lot of confusion?  I know there are not many of us running older versions of Mac OS X left, but there are a few!