arduino/Arduino #10063: Add SparkFun Qwiic Humidity AHT20 Arduino Library

pmakin-png, 2020-04-17:
Hello,

Please add our AHT20 Arduino library to the library manager. Here is the link:

pmakin-png, 2020-04-17:
https://github.com/sparkfun/SparkFun_Qwiic_Humidity_AHT20_Arduino_Library

Thanks!

cmaglie, 2020-04-25:
Hi @pmakin-png 

I've added the library to the indexer but there is a problem:

```
JOB 2939 - 2020/04/25 18:13:33 Scraping https://github.com/sparkfun/SparkFun_Qwiic_Humidity_AHT20_Arduino_Library.git
JOB 2939 - 2020/04/25 18:13:33 Checking out tag: 1.0.0
JOB 2939 - 2020/04/25 18:13:33 Error syncing library: Error generating library from repo: Invalid character(s) found in major number "v1"
```

the `version` field in library.properties contains a `v` as a prefix of the version:
```
version=v1.0.0
```
it must be:
```
version=1.0.0
```
to be semver compliant.

To fix the above:
1. change the version field to 1.0.0 in library.properties
2. commit and push
3. delete and re-do the git tag/release

once done the above, the library will be automatically published within an hour.
If this doesn't happen please comment here so we can give you futher assistance.

closed 2020-04-25 by cmaglie