arduino/Arduino #10071: Please add new library for max7219

JemRF, 2020-04-20:
Please add the max7219 library to the library manager:

https://github.com/JemRF/max7219

This library allows user to easily add the max7219 7 segment, 8 digit display to their Arduino projects..

per1234, 2020-04-21:
Hi @JemRF. There are currently a couple of problems that will block the addition of your library to the Arduino Library Manager index:

---
In order to be added to the Library Manager index your library must contain a library.properties file. The format of this file is documented here:
https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format

---
After adding the library.properties file, you will need to create a new tag/release. Make sure that the `version` value matches the tag of the release that contains it.

Once you have done this please comment here so we know we can proceed.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

JemRF, 2020-04-21:
Thanks for that information. I have completed those steps.

JemRF, 2020-04-25:
> Hi @JemRF. There are currently a couple of problems that will block the addition of your library to the Arduino Library Manager index:
>
> In order to be added to the Library Manager index your library [...]

Thanks for that information. I have completed those steps.

cmaglie, 2020-04-25:
Done.

closed 2020-04-25 by cmaglie