arduino/Arduino #10074: AsyncIOT

SanUSB, 2020-04-21:
[AsyncIOT-master.zip](https://github.com/arduino/Arduino/files/4507929/AsyncIOT-master.zip)
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.

SanUSB, 2020-04-21:
AsyncIOT provides:

an UNIVERSAL HMI (human-machine interface) environment sketch for ESP32 and ESP8266 in order to interact with processed inputs and outputs using any browser through HTML pages. As an example, the input parameters values are converted to integers, added, multiplied and the results are shown in the output parameters. The number of input and output parameters is also set by the user.

This environment also makes it possible to update the firmware .bin via WiFi OTA (Over the Air). To generate a .bin file from your sketch, go to Sketch menu of the Arduino IDE > Export compiled Binary.

Sandro Jucá
http://lattes.cnpq.br/0543232182796499

per1234, 2020-04-21:
Hi @SanUSB. The purpose of this issue report is not clear. Are you requesting that your library be added to the Arduino Library Manager index?

You can find information about this in the FAQ:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

SanUSB, 2020-04-21:
Hi, per1234. Yes, I request that my library be added to the Arduino Library Manager index. 
Is there something wrong with the proposal?

Thank you.

per1234, 2020-04-21:
The first problem with your proposal is that you didn't post the URL of your library's repository. The Arduino Library Manager index is populated by an indexer job that runs periodically and picks up all tags from the repository URLs in the index. We have no use for a .zip file.

The second problem is that your repository has no tags for the indexer job to pick up.

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

SanUSB, 2020-04-21:
Please to consider: 

https://github.com/SanUSB/AsyncIOT

https://github.com/SanUSB/AsyncIOT/blob/master/library.properties

Is that correct?

Thank You

per1234, 2020-04-21:
No, it's not correct. You forgot to fix one of the two problems I mentioned in my last reply:
>The second problem is that your repository has no tags for the indexer job to pick up.
>
>You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

per1234, 2020-08-04:
Closing due to the library still not being compliant.

@SanUSB, if you ever get around to making a release, comment here and I'll be happy to reopen this issue.

closed 2020-08-04 by per1234