arduino/Arduino #10078: Please add library SolarCharger

hectorespert, 2020-04-22:
Please add library SolarCharger https://github.com/hectorespert/SolarCharger

per1234, 2020-04-22:
Hi @hectorespert. Tags which contain a `.development` file will be ignored by the Arduino Library Manager indexer. Since your library's only tag contains this file:
https://github.com/hectorespert/SolarCharger/blob/master/.development
it means that your library can't be published.

After removing the .development file, you will need to create a new tag/release or redo the existing one. Make sure that the `version` value matches the tag of the release that contains it.

Once you have done this please comment here so we know we can proceed.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

hectorespert, 2020-04-22:
@per1234 Fixed, sorry for the inconvenience

cmaglie, 2020-04-25:
hi @hectorespert 

in your library.properties is still missing the `author` field:
```
JOB 2948 - 2020/04/25 18:13:34 Scraping https://github.com/hectorespert/SolarCharger.git
JOB 2948 - 2020/04/25 18:13:36 Checking out tag: 1.0.0
JOB 2948 - 2020/04/25 18:13:36 Error syncing library: Error generating library from repo: 'author' field must be defined
```

to fix this:
1. add the `author` field to library.properties
2. commit & push
3. delete and redo the git release tag.

since the library has been added to the indexer, once you do the above steps the library should be published automatically within an hour.

I'm going forward and closing this issue, if you need more assistance feel free to comment further.

closed 2020-04-25 by cmaglie