arduino/Arduino #10082: #error unknown CPU

RobDouglas85, 2020-04-24:
Hi All, 

I am new to the Nano 33 IoT and web editor game so I don't quite understand why a sketch that compiles in the IDE and loads onto the board fine gives me the following error in the Web Editor.  I probably missed something really rudimentary but I can't put my finger on it.  

Does the Arduino Web editor use a different compiler?  In the IDE i have AVRISP mkII.  Does that make a difference?

Any thoughts would be helpful.  Thanks!

*********
Error Message:

Using library digital_infrared_temperature_sensor_mlx90615_1_0_0 at version 1.0.0 in folder: /home/builder/opt/libraries/latest/digital_infrared_temperature_sensor_mlx90615_1_0_0

In file included from /home/builder/opt/libraries/latest/digital_infrared_temperature_sensor_mlx90615_1_0_0/MLX90615.h:5:0,

from /tmp/922428008/wifi_bmp180_LED_IRDist_IRTemp/wifi_bmp180_LED_IRDist_IRTemp.ino:30:

/home/builder/opt/libraries/latest/digital_infrared_temperature_sensor_mlx90615_1_0_0/I2cMaster.h:138:4: error: #error unknown CPU
#error unknown CPU
^~~~~
/tmp/922428008/wifi_bmp180_LED_IRDist_IRTemp/wifi_bmp180_LED_IRDist_IRTemp.ino:34:10: fatal error: SFE_BMP180.h: No such file or directory
#include <SFE_BMP180.h>
^~~~~~~~~~~~~~
compilation terminated.
exit status 1

*******
Code (it's a mess, but it works):
-----------------
#include <Wire.h>

// Digital Infrared Temperature Sensor MLX90615 - Version: Latest 
#include <MLX90615.h>
MLX90615 mlx = MLX90615();

#include <SFE_BMP180.h>
SFE_BMP180 pressure;
#define ALTITUDE '0'// in meters

int red_light_value;
int green_light_value;
int blue_light_value;

int red_light_pin= 2;
int green_light_pin = 3;
int blue_light_pin = 4;

void setup() {
  Serial.begin(115200);
  while (!Serial) {
    ; 
  }

  if (WiFi.status() == WL_NO_MODULE) {
    Serial.println("Communication with WiFi module failed!");
    while (true);
  }

  String fv = WiFi.firmwareVersion();
  if (fv < WIFI_FIRMWARE_LATEST_VERSION) {
    Serial.println("Please upgrade the firmware");
  }

  while (status != WL_CONNECTED) {
    Serial.print("Attempting to connect to WPA SSID: ");
    Serial.println(ssid);
    status = WiFi.begin(ssid, pass);

    delay(10000);
  }

  Serial.print("You're connected to the network");
  printCurrentNet();
  printWifiData();

Serial.println("REBOOT");
  
  Serial.println("Melexis MLX90615 infra-red temperature sensor test");
  mlx.begin();
  Serial.print("Sensor ID number = ");
  Serial.println(mlx.get_id(), HEX);
  
  if (pressure.begin())
    Serial.println("BMP180 init success");
  else
  {

    Serial.println("BMP180 init fail\n\n");
    while(1); 
  }
  pinMode(red_light_pin, OUTPUT);
  pinMode(green_light_pin, OUTPUT);
  pinMode(blue_light_pin, OUTPUT);
  
}

void loop() {
  // check the network connection once every 10 seconds:
  delay(10000);
  printCurrentNet();

  Serial.print("IR1 Ambient = ");
  Serial.print(mlx.get_ambient_temp());
  Serial.println(" *C\t"); 
  Serial.print("IR1 Object = ");
  Serial.print(mlx.get_object_temp());
  Serial.println(" *C");

  char status;
  double T,P,p0,a;

  // Loop here getting pressure readings every 10 seconds.

  Serial.println();
  Serial.print("provided altitude: ");
  Serial.print(ALTITUDE,0);
  Serial.print(" meters, ");
  Serial.print(ALTITUDE*3.28084,0);
  Serial.println(" feet");
  
  
  status = pressure.startTemperature();
  if (status != 0)
  {
      delay(status);

      status = pressure.getTemperature(T);
    if (status != 0)
    {
          Serial.print("temperature: ");
      Serial.print(T,2);
      Serial.print(" deg C, ");
      Serial.print((9.0/5.0)*T+32.0,2);
      Serial.println(" deg F");
      
            status = pressure.startPressure(3);
      if (status != 0)
      {
          delay(status);

       

        status = pressure.getPressure(P,T);
        if (status != 0)
        {
         
          Serial.print("absolute pressure: ");
          Serial.print(P,2);
          Serial.print(" mb, ");
          Serial.print(P*0.0295333727,2);
          Serial.println(" inHg");

        
          p0 = pressure.sealevel(P,ALTITUDE);
          Serial.print("relative (sea-level) pressure: ");
          Serial.print(p0,2);
          Serial.print(" mb, ");
          Serial.print(p0*0.0295333727,2);
          Serial.println(" inHg");

        
          // Parameters: P = absolute pressure in mb, p0 = baseline pressure in mb.
          // Result: a = altitude in m.

          a = pressure.altitude(P,p0);
          Serial.print("computed altitude: ");
          Serial.print(a,0);
          Serial.print(" meters, ");
          Serial.print(a*3.28084,0);
          Serial.println(" feet");
        }
        else Serial.println("error retrieving pressure measurement\n");
      }
      else Serial.println("error starting pressure measurement\n");
    }
    else Serial.println("error retrieving temperature measurement\n");
  }
  else Serial.println("error starting temperature measurement\n");

//  RGB_color(255, 0, 0); // Red
//  RGB_color(0, 255, 0); // Green
//  RGB_color(0, 0, 255); // Blue
//  RGB_color(255, 255, 125); // Raspberry
//  RGB_color(0, 255, 255); // Cyan
//  RGB_color(255, 0, 255); // Magenta
//  RGB_color(255, 255, 0); // Yellow
//  RGB_color(0, 0, 0); // White
  
if  (T <=20.0) {
    RGB_color(255, 0, 0); // Red
}
else if (T <=21) {
  RGB_color(255, 255, 125); // Raspberry
}
else if (T <=22) {
  RGB_color(0, 255, 255); // Cyan
}
else if (T <=23) {
  RGB_color(255, 0, 255); // Magenta
}
else if (T <=24) {
  RGB_color(255, 255, 0); // Yellow
}
else if (T <=25) {
    RGB_color(0, 255, 0); // Green
}
else if (T <=26.0) {
    RGB_color(0, 0, 255); // Blue
}
else {
    RGB_color(255, 255, 125); // Raspberry
}
  Serial.println();
  Serial.println("======");
  Serial.println();
  delay(1000);
  

}

void printWifiData() {

  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);
  Serial.println(ip);

  byte mac[6];
  WiFi.macAddress(mac);
  Serial.print("MAC address: ");
  printMacAddress(mac);
}

void printCurrentNet() {
 
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

 
  byte bssid[6];
  WiFi.BSSID(bssid);
  Serial.print("BSSID: ");
  printMacAddress(bssid);

  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.println(rssi);

  byte encryption = WiFi.encryptionType();
  Serial.print("Encryption Type:");
  Serial.println(encryption, HEX);
  Serial.println();
}
void RGB_color(int red_light_value, int green_light_value, int blue_light_value)
 {
  analogWrite(red_light_pin, red_light_value);
  analogWrite(green_light_pin, green_light_value);
  analogWrite(blue_light_pin, blue_light_value);
}
void printMacAddress(byte mac[]) {
  for (int i = 5; i >= 0; i--) {
    if (mac[i] < 16) {
      Serial.print("0");
    }
    Serial.print(mac[i], HEX);
    if (i > 0) {
      Serial.print(":");
    }
  }
  Serial.println();
}

per1234, 2020-04-24:
Hi @RobDouglas85. I'm closing this issue because the correct place to report issues related to the Arduino Web Editor is the Arduino forum's **Create > Editor** board:
http://forum.arduino.cc/index.php?board=101.0
If you haven't done so already, please make a post there. I know exactly what the problem is and can definitely help you out over there. Thanks!

closed 2020-04-24 by per1234

RobDouglas85, 2020-04-24:
Ok, thanks.