arduino/Arduino #10083: IDE failure: Java exception

supuflounder, 2020-04-24:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.
`

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\admin\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\admin\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\admin\Documents\Arduino\libraries -fqbn=esp32:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=none,PartitionScheme=default -vid-pid=10C4_EA60 -ide-version=10812 -build-path C:\Users\admin\AppData\Local\Temp\arduino_build_540982 -warnings=more -build-cache C:\Users\admin\AppData\Local\Temp\arduino_cache_208070 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=C:\Users\admin\Documents\ArduinoData\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\admin\Documents\ArduinoData\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\admin\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\admin\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\admin\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\admin\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose C:\Users\admin\OneDrive\Projects\NeoPixel Thermometer\Source\NeoPixel_Thermometer_V1\NeoPixel_Thermometer_V1.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\admin\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\admin\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\admin\Documents\Arduino\libraries -fqbn=esp32:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=none,PartitionScheme=default -vid-pid=10C4_EA60 -ide-version=10812 -build-path C:\Users\admin\AppData\Local\Temp\arduino_build_540982 -warnings=more -build-cache C:\Users\admin\AppData\Local\Temp\arduino_cache_208070 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=C:\Users\admin\Documents\ArduinoData\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\admin\Documents\ArduinoData\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\admin\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\admin\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\admin\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\admin\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose C:\Users\admin\OneDrive\Projects\NeoPixel Thermometer\Source\NeoPixel_Thermometer_V1\NeoPixel_Thermometer_V1.ino
Using board 'featheresp32' from platform in folder: C:\Users\admin\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4
Using core 'esp32' from platform in folder: C:\Users\admin\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4
Detecting libraries used...
"C:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_event" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_client" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_server" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_https_ota" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ethernet" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/expat" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fatfs" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freemodbus" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freertos" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/heap" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/idf_test" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/jsmn" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/json" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/libsodium" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/log" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/lwip" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mbedtls" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mdns" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/micro-ecc" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mqtt" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/newlib" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nghttp" "-IC:\\Users\\admin\\Documents\\ArduinoData\\packages\\esp [... cut ...]
	at processing.app.Editor.statusError(Editor.java:2533)
	at processing.app.Editor$BuildHandler.run(Editor.java:1632)
	at java.lang.Thread.run(Thread.java:748)
`
I have a zip file of the source that caused the failure (the ton of errors is because I am halfway through a major refactoring of the code), but it should still not take a Java exception.  I did not find how to attach this file to this message, but if you  contact me, I can reply with the attachment. Also, I thought the "insert code" would put the text into a scrollable box.

facchinm, 2020-04-27:
It looks like the IDE is trying to highlight the error line but for some reason the line contains a token that is not handled correctly.
If the IDE doesn't stop working I would consider it a minor issue and leave it as is. Otherwise we can investigate a bit more

supuflounder, 2020-04-28:
The IDE continues to run.

facchinm, 2020-04-28:
OK, thanks for reporting. I'd close it as wontfix, tagging appropriately to make sure we can find it again if it gets reported again.

closed 2020-04-28 by facchinm