arduino/Arduino #10100: TFMini-Plus-I2C Arduino Library

budryerson, 2020-04-26:
Hello again,

The `0.1.0` tag was deleted on 19FEB20, and many Sundays have passed since then.
The Arduino library is still showing version `0.4.0` as the latest version.
That is not correct. Version `0.2.2` is the latest,
I could use your advice, but I would prefer your assistance because in this instance it appears that the indexer is not doing its job correctly.

Thank you very much,

Bud Ryerson
San Francsico
<hr>
per1234 commented on Feb 19

The reason is the version field in library.properties was set to 0.4.0 when the 0.1.0 tag was created:
https://github.com/budryerson/TFMini-Plus-I2C/blob/0.1.0/library.properties#L2

You can fix this by deleting the 0.1.0 tag:
https://github.com/budryerson/TFMini-Plus-I2C/releases/tag/0.1.0
deleted tags are processed by the indexer job every Sunday, so the 0.4.0 version will still be available in Library Manager until then.
@per1234 per1234 closed this on Feb 19
<hr>
Original request 18FEB19

Hi folks,

I am so sorry to have to bother you with this.

A version 0.4.0 of the TFMini-Plus-I2C library shows up in the Arduino Library Manager.
This is wrong. There is no version 0.4.0. I don't know how this happened
and I do not know how to fix this, or how to make it go away.

The correct and current version is v0.2.2.
Please help me to make v0.4.0 go away.
Can you fix this? Please?

Again, my apologies.

Bud Ryerson
San Francisco

per1234, 2020-04-27:
The tag is still there. Perhaps you deleted the [GitHub release](https://help.github.com/en/github/administering-a-repository/managing-releases-in-a-repository), but the [Git tag](https://git-scm.com/book/en/v2/Git-Basics-Tagging) was never deleted.

You need to delete the `0.1.0` tag. You can do this by going to this page:
https://github.com/budryerson/TFMini-Plus-I2C/releases/tag/0.1.0
and clicking the "Delete" button.

After that, I believe you will need to wait for the next round of manually processing Library Manager requests, due to a recent change to the system for distributing the Library Manager index file that is currently preventing releases from being automatically removed from the index.

budryerson, 2020-04-27:
You are correct.  _Now_ I have deleted the tag.
I apologize for my inexperience.  I'd like to keep this open until next Sunday to be sure.
Thank you for your assistance.
Bud

cmaglie, 2020-05-04:
removed 0.4.0

closed 2020-05-04 by cmaglie