arduino/Arduino #10108: [Library Manager] please add DBS-Lib 1.0.1

DBSStore, 2020-04-28:
https://github.com/DBSStore/DBS_Lib/releases/tag/1.0.1
https://github.com/DBSStore/DBS_Lib
I updated the library from 1.0.0 to 1.0.1 to add various dimensions.

per1234, 2020-04-28:
Hi @DBSStore. There is no need to open an issue for every release of your library. After the initial addition of the library, the indexer job periodically checks each of the libraries in the index and picks up any new release/tag that meets [the requirements](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ).

I notice there is a problem with your 1.0.1 release which will prevent it from being picked up by the indexer. You didn't update the `version` value in library.properties before creating the tag/release. The Library Manager uses the library.properties `version` value for versioning, so any tag that doesn't have a unique `version` value in library.properties is ignored. At the time of your 1.0.1 release, the `version` value was still set to `1.0.0`:
https://github.com/DBSStore/DBS_Lib/blob/1.0.1/library.properties#L2
I see you have since updated the `version` value, so you will only need to redo the existing 1.0.1 release, or else create a new release.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

closed 2020-04-28 by per1234

DBSStore, 2020-04-28:
I solved