arduino/Arduino #10113: Nano Pinout on Website shows PIN4 as PWM

Palmstroemen, 2020-04-29:
In the documentation for NANO on the Website [https://store.arduino.cc/arduino-nano](https://store.arduino.cc/arduino-nano) in the tab "Documentation" there is a pinout diagram.
There Pin4 is shown as PWM while Pin3 is shown as notPWM.
Mostly all other pinout diagrams found in the web show it vice versa. Pin3 is PWM and Pin4 is notPWM. They also often show Pin9, 10, 11 to be PWM in contrast to the official document mentioned here.

per1234, 2020-05-06:
You're right that PWM pins 9, 10, and 11 are also PWM pins, but not marked as such on the diagram.

FunzioneFabio, 2020-07-01:
@Palmstroemen Thank you so much for reporting this, the pinout is now corrected and uploaded on the website!
https://content.arduino.cc/assets/Pinout-NANO_latest.pdf

closed 2020-07-01 by kengdahl