arduino/Arduino #10125: Fix for SAMD51 upload failures

tedholler, 2020-05-01:
Issue:
  Uploading a sketch to a SAMD51 device (Adafruit feather M4 Express, Adafruit M Metro, Alorium EVO) will intermittently fail with a "SAM-BA operation failed" error.

Platform: MacOSx and Linux

File:
  arduino-core/src/cc/arduino/packages/uploaders/SerialUploader.java

Requested Change:

     **_int maxTries
     maxTries = (board_is_SAMD51) ? 3 : 1;
     for(int tries=0; tries < maxTries; tries++) {
        if (verbose)
           System.out.println(
             I18n.format(tr("Forcing reset using 1200bps try={0} open/close on port {1}"), tries, userSelectedUploadPort));
        Serial.touchForCDCReset(userSelectedUploadPort);
        if(board_is_SAMD51) Thread.sleep(5000);
     }_**

Details:
  The BOSSA's Samba.cpp and PosixSerialPort.cpp code will timeout while doing a byteRead
   from the SAMD51 after the Samba code writes a command to SAMD51

  The ardiuno example sketch "BarometricPressureWebServer" is the sketch I used while 
   debugging this issue

  I connected a logic analyzer up to the USB port and captured the USB data being 
  sent/received.
  I can see the USB transfer for the byteRead reference and the transfer looks to be correct
  and the SAMD51 did return the correct data.
  But right after this I see in the USB data transfer that the HOST switched the baud-rate from 
  9600 bps to either 57600 or 921600 bps
  I'm unable to tell if the baud-rate change caused the readByte request to timeout or a result 
  of the OS handling of the timeout.

  This failure happen when using a Ubuntu Linux system as well as a MacOSx system
  Linux: 5.3.0-42-generic
  MacOS: Darwin Kernal Version 18.7.0
  On the Linux system the upload would fail every other time
  On the MacOS system the upload would fail around 5% of the time

  While debugging this issue I notice that on the Linux system the 
  "EVOM51BOOT"/"FEATHERBOOT" icon is being display
  on the desktop when the Sketch is upload completely.
  When the icon isn't displayed the uploading of the sketch will fail.
  This made me think that it might be an initialization issue

  So I added the code above (I didn't use the board_is_SAMD51 variable as I'm not sure where
  to get that information) to force the SAMD51 reset (open/close port at 1200bps) 3 times 
  before starting the actuall upload.

  I have not seen any upload failures to the SAMD51 device when using the updated code.

  You can contact me at holler@superiontech.com
  Ted Holler

facchinm, 2020-05-04:
Hi @tedholler ,
we are transitioning our low lever effort to https://github.com/arduino/arduino-cli , which will act as a backend for future IDE releases.
Would you mind checking if that implementation just work on your systems? Also, on LInux, make sure `modemmanager` is not installed (it can harm the upload by locking the serial port for some seconds after it appears)

tedholler, 2020-05-05:
Hi,  Martino!  Thanks for getting back to me about this issue.

I got the latest copy of arduino-cli for both the Linux and MacOSx

I did uninstalled the modemmanager on the Linux system.

Linux:
   So far I have not seen any issues uploading sketches to the Adafruit feather M4 Express or the Alorium EVO boards

MacOS:
   I can only get a sketch to upload without failures right after pressing the reset button on the board (both boards have the same behavior)
   Trying to upload a sketch after the 1st upload the upload fails due to “SAM-BA operation failed”