arduino/Arduino #10126: [Library Manager] Please add FRAM_I2C library

RobTillaart, 2020-05-01:
Please add https://github.com/RobTillaart/FRAM_I2C
Thanks

per1234, 2020-05-02:
Hi @RobTillaart. Each library added to the Library Manager index must have a unique library.properties `name` value (regardless of case). The `name` value "FRAM" used in your library has already been taken:
https://github.com/Industruino/FRAM

If you would still like your library added to the index, you'll need to change the `name` value to a unique name, then either update the `version` value in library.properties and make a new release/tag or else redo the existing tag.

After that is done, please comment here to let us know we can proceed. Thanks!

RobTillaart, 2020-05-02:
Sorry, my mistake,
changed the name field  to FRAM_I2C and released the library again

cmaglie, 2020-05-04:
Done.

closed 2020-05-04 by cmaglie