arduino/Arduino #10128: [Library manager] Add MQ131 driver

ostaquet, 2020-05-01:
Hi
Could you add this driver for the MQ131 sensor (ozone sensor) :
https://github.com/ostaquet/Arduino-MQ131-driver/releases/tag/1.3.0
Thanks!

cmaglie, 2020-05-04:
HI @ostaquet 

your library name contains characters that are not allowed:

```
MQ131 gas sensor driver (Ozone/O3)
```
in particular `(` and `/` doesn't work well. Please rename your library:

1) edit library.properties and change the `name` field
2) push & commit
3) delete and redo the 1.3.0 tag

I suggest you `MQ131` as name (the words `ozone` and `sensor` are already in the description that is searchable through the IDE GUI).
Please comment here when you are done so we can proceed to add the library.

ostaquet, 2020-05-04:
I missed that one. Sorry.

New release tag as 1.3.1 (https://github.com/ostaquet/Arduino-MQ131-driver/releases/tag/1.3.1)

cmaglie, 2020-05-09:
Done.

closed 2020-05-09 by cmaglie