arduino/Arduino #10130: Hourly Build 1.8.13 2020/04/27 05:25 startup slow

jpmeijers, 2020-05-02:
Version: 1.8.13 Hourly Build 2020/04/27 05:25
OS: Ubuntu 20.04

When starting the IDE the loading screen shows immediately, but it takes almost a minute before the loading screen says "Initializing Packages" and "Starting". In the terminal I only see the following message before the wait starts.

```
$ arduino
Picked up JAVA_TOOL_OPTIONS: 
Set log4j store directory /home/jpmeijers/.arduino15
Gtk-Message: 14:03:48.223: Failed to load module "canberra-gtk-module"
```

mitsake, 2020-05-14:
I had this issue with a fresh Ubuntu 20.04 install and Arduino 1.8.12. It also occurred with beta 1.9.0, build 119. Updating to the hourly build dated 13 May 2020 2:35:6 GMT appears to have resolved the issue for me.

matthijskooijman, 2020-05-14:
Hm, looking through the recently merged pullrequests, I don't see anything to explain a speedup.

@jpmeijers, could you see if the latest hourly build also fixes this slowdown for you?

jpmeijers, 2020-05-14:
Tested using nightly build 12 May 2020, Linux 64 bit. Still takes 26 seconds to do anything after launch. Splash screen shown as in the screenshot attached during this 26 seconds.

[image]

jpmeijers, 2020-05-18:
I was running into a different issue today which prompted me to first downgrade to 1.8.12, and then to 1.8.11. Both these releases also have a 27 second stall during startup.

matthijskooijman, 2020-05-18:
I suspect this might be caused by #10214. If so, the delay depends on the selected board: For boards with more "option" menus, the startup delay is longer. So if this problem disappears when some simple board like the Uno is selected, this could be the same problem.

jpmeijers, 2020-05-18:
I don't think so. I did the following experiment:

* Open Arduino
* Set board to Arduino Uno
* Open Blink example
* Close all Arduino windows except Blink
* Close Arduino
* Open Arduino
* Blink opens and Arduino Uno is selected

Startup still hung for 26/27 seconds

mitsake, 2020-05-27:
Sorry I've come back to this so late. I made the error of assuming that the change I made was what fixed the issue, and it turns out it wasn't that at all - every version of the IDE I've tested since then has started up fine, so it must be to do with something else. I might try a VM with a fresh install of Ubuntu and see if I can replicate it.

jpmeijers, 2020-06-25:
I think it was an issue with Ubuntu, because at some point it started launching without the delay. I've tested now with 1.8.11 and 1.8.13 and they are both fine. So closing the issue for now.

closed 2020-06-25 by jpmeijers

Erriez, 2020-08-28:
I'd like to reopen this issue, because the solution is missing.

Several Arduino IDE's versions are started with the 26/27 seconds delay on a fresh Ubuntu 20.04.1 installation including latest updates.

I performed the following experiments without success:
- Versions 1.8.12, 1.8.13 and nightly.
- Tested versions above with `~/.arduino15` directory removed.
- Set board to Uno / open Blink example.

No error messages on the console:
```
$ arduino
Picked up JAVA_TOOL_OPTIONS: 
Set log4j store directory /home/erwin/.arduino15
```

To resolve `Failed to load module "canberra-gtk-module"` (has no effect on the startup delay):
```
sudo apt install libcanberra-gtk-module libcanberra-gtk3-module
```

Can someone advise me what to do next?

Erriez, 2020-08-28:
The Arduino IDE v1.8.13 starts within 1 second on a fresh Ubuntu 18.04.3 with all updates installed. So I decided to downgrade the OS as workaround until a fix is available.

Can someone reproduce this extreme startup delay on Ubuntu 20.04.1? A virtual machine like VirtualBox can be used.

carlesoriol, 2020-09-14:
I'm on the same bug also in ubuntu 20.04.129 seconds to boot (i5 machine with ssd)

Erriez, 2020-10-09:
@carlesoriol Now it works for me after installing the latest Ubuntu 20.04 updates: The IDE starts within 1 second as expected.