arduino/Arduino #10141: Syntax IMU.readAcceleration(x,y,z) misplaced

alejandror, 2020-05-04:
I think there is an error in description  of read magnetic field function
 Syntax
IMU.readAcceleration(x,y,z) missplaced in for https://www.arduino.cc/en/Reference/ArduinoLSM9DS1readMagneticField
correct sintax should be: IMU.readMagneticField(x, y, z);
sorry if wrong place, or already issued

SuperGops7, 2020-05-16:
I would love to fix this, but I'm not able to find the Arduino LSM9DS3 library in the references-en repo.

per1234, 2020-05-16:
Hi @SuperGops7. Thanks for your interest in improving Arduino's documentation! The [arduino/reference-en repository](https://github.com/arduino/reference-en) only contains the content for the [Arduino Language Reference](https://www.arduino.cc/reference/en/). The content for the [Arduino library reference pages](https://www.arduino.cc/en/Reference/Libraries) is not hosted in a public repository, so there is no way to submit pull requests for modification to that content. If you find issues with the library reference pages, you are welcome to submit an issue. It's helpful to provide the specific text you would suggest in order to make implementing the change as close as possible to a quick copy/paste job for the documentation maintainers.

per1234, 2021-03-28:
Thanks so much for reporting this @alejandror. The error has now been corrected.

closed 2021-03-28 by per1234