arduino/Arduino #10155: [Library Manager] AllThingsTalk LTE-M SDK

epiller, 2020-05-06:
Hello! Could you please add our SDK to the Library Manager? 
If it makes the verification process any faster, we already have two libraries published in Library Manager (_AllThingsTalk WiFi SDK_ and _AllThingsTalk LoRaWAN SDK_).

### AllThingsTalk LTE-M SDK: https://github.com/allthingstalk/arduino-ltem-sdk

cmaglie, 2020-05-09:
Done.

closed 2020-05-09 by cmaglie

epiller, 2020-06-19:
@cmaglie I've published a major new version of [AllThingsTalk LTE-M SDK](https://github.com/allthingstalk/arduino-ltem-sdk) today but I noticed that neither this nor the previous version are being picked up.  
We are following all [guidelines](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-is-the-library-list-generated) for Arduino Library Manager. What gives?

per1234, 2020-06-19:
Hi @epiller. The reason the 1.0.1 release was not picked up is because you didn't update the `version` value in library.properties before making the release:
https://github.com/allthingstalk/arduino-ltem-sdk/blob/1.0.1/library.properties#L2
The Library Manager indexer works from the library.properties `version` value, not the tag name, so the index already had a `1.0` version of the library.

As for your 2.0 release. It is fine and already available from Library Manager for me:
[image]
however, it might take some time for the updated index to propogate through the cache to become available to everyone, so if it's not available to you yet, check back in a little while and I think it will be.

epiller, 2020-06-19:
You were absolutely right! Thank you so much for the quick response - everything is fine now.