arduino/Arduino #10158: [Library Manager] New version not showing up and version sorted incorrectly

matthijskooijman, 2020-05-06:
I have this library added to the library manager: https://github.com/matthijskooijman/arduino-lmic

There's two problems:
 - The latest version does not show up in the IDE library manager. It seems to have a correct tag, and the version number in library.properties seems also correct: https://github.com/matthijskooijman/arduino-lmic/tree/1.5.0%2Barduino-3 Any suggestions on what could be wrong? There are a few forked repositories around, I could imagine maybe a fork was added? Is the original repo url for a library shown somewhere?
 - The version numbers are sorted in reverse. I suspect this is something with the + used in the version number? 1.5.0+arduino-0 is always shown as the most recent version, which causes people to install an old version instead of the latest version. This is probably something in the IDE source which I could dig into myself, just haven't done so yet...

per1234, 2020-05-06:
>The latest version does not show up in the IDE library manager.

I notice the name was changed between the the [1.5.0+arduino-2](https://github.com/matthijskooijman/arduino-lmic/blob/1.5.0%2Barduino-2/library.properties#L1) and [1.5.0+arduino-3](https://github.com/matthijskooijman/arduino-lmic/blob/1.5.0%2Barduino-3/library.properties#L1) tags.

My "black box" understanding of the indexer is that it should still pick up new tags after the name change, but they will be pinned to the name at the time of the original addition of the library to the index until a request to update the name is submitted, so I wouldn't expect the changed `name` value to cause the tag to be rejected.

>Is the original repo url for a library shown somewhere?

It's shown under the `repository` key in [the index](http://downloads.arduino.cc/libraries/library_index.json).

cmaglie, 2020-05-09:
The problem is that you changed library name, this is not allowed. The indexer will ignore all releases with the wrong name.

```
JOB 512 - 2020/05/09 17:24:52 Scraping https://github.com/matthijskooijman/arduino-lmic.git
JOB 512 - 2020/05/09 17:24:53 Checking out tag: 1.5.0+arduino-0
JOB 512 - 2020/05/09 17:24:53 Release IBM LMIC framework:1.5.0+arduino-0 already loaded, skipping
JOB 512 - 2020/05/09 17:24:53 Checking out tag: 1.5.0+arduino-1
JOB 512 - 2020/05/09 17:24:53 Release IBM LMIC framework:1.5.0+arduino-1 already loaded, skipping
JOB 512 - 2020/05/09 17:24:53 Checking out tag: 1.5.0+arduino-2
JOB 512 - 2020/05/09 17:24:53 Release IBM LMIC framework:1.5.0+arduino-2 already loaded, skipping
JOB 512 - 2020/05/09 17:24:53 Checking out tag: 1.5.0+arduino-3
JOB 512 - 2020/05/09 17:24:53 Release LMIC-Arduino:1.5.0+arduino-3 has wrong library name, should be IBM LMIC framework
```

If you want I can rename the library in the library manager, but the previous releases will be lost (because they will be ignored after the change of name).

Other suggestions that comes from my personal taste but I would like share anyway :smile::
- I would not put `Arduino` in the library name becuase it's redundant.
- I would not use sub/build versions (`x.x.x+something` or `x.x.x-some+thingelse`) becuase I see they cause a lot of confusion.

cmaglie, 2020-05-09:
About the version numbers I've looked at the [SEMVER](https://semver.org/#spec-item-10) specification about "build metadata" (i.e. the identfier followed by the `+` sign), it says:

> 10. Build metadata MAY be denoted by appending a plus sign and a series of dot separated identifiers immediately following the patch or pre-release version. Identifiers MUST comprise only ASCII alphanumerics and hyphen [0-9A-Za-z-]. Identifiers MUST NOT be empty. **Build metadata MUST be ignored when determining version precedence. Thus two versions that differ only in the build metadata, have the same precedence**. Examples: 1.0.0-alpha+001, 1.0.0+20130313144700, 1.0.0-beta+exp.sha.5114f85.

So if I read it correctly the precedence between:

- 1.5.0+arduino-0
- 1.5.0+arduino-1
- 1.5.0+arduino-2

is undefined and the IDE theoretically may choose any of those and still be correct...

marcobrianza, 2020-09-15:
Hi, @matthijskooijman this issue is preventing to get the latest release, the previous one do not compile anymore with the updated Arduino IDE. Ciao Marco

matthijskooijman, 2020-09-27:
Woah, seems I totally missed all the replies on this issue back when they were posted, not sure how that happened. Thanks for you quick replies, apologies for only following up now.

>    I would not put Arduino in the library name becuase it's redundant.
>    I would not use sub/build versions (x.x.x+something or x.x.x-some+thingelse) becuase I see they cause a lot of confusion.

The reason for both is actually the same: This is an Arduino port of an upstream library, which I wanted to emphasize with the name. Especially since some development has happened in the Arduino port and upstream sort of halted development, it no longer seemed proper to use the IBM name. But I can probably rename it back for now, it's not a really big deal.

Similarly, for the version numbers, I didn't want to change the upstream version number, but use them verbatim and add an arduino-specific addition, so I would not conflict with new upstream releases. Given that upstream is totally dead and has been pretty much since the start, I guess I can just bump to 1.5.1 for now.

To further complicate matters, I've not really developed much on this library, and there is another fork (MCCI LMIC) which did do more development and did bump versions, so I didn't want to conflict with that for now.

Anyway, thanks for the feedback, I think I know enough to sort out my own mess now :-)

closed 2020-09-27 by matthijskooijman