arduino/Arduino #10162: Enhancement request for the IDE: Favourites in the list of boards.

rin67630, 2020-05-06:
The Arduino IDE is victim of its success.
Now the list of boards supported spans over several pages and is reset every time the mouse accidentally leaves the long list.
Mainly users are only actively interested in a couple of them.
IMHO it would be really useful, if the IDE could present the most used "favorites" at the very beginning of the list.
Thank you for considering.

per1234, 2020-05-06:
Hi @rin67630. This feature is already implemented in https://github.com/arduino/Arduino/pull/8607

You can try it out in the beta build of the Arduino IDE:
https://www.arduino.cc/en/Main/Software#beta

rin67630, 2020-05-07:
Was a kind of mind transmission? I feel perfectly at home at #8607
Closing that issue with thanks.

closed 2020-05-07 by rin67630