arduino/Arduino #10171: The Arduino-builder compiler doesn't work 1.8.12 linux 64

HennieP12, 2020-05-07:
Arduino-builder doesn't work linux 64 bit, in 1.8.12.<br>Ran arduino-builder from terminal, segmentation fault.<br>Arduino itself (1.8.12) returned with error code 139. Thanks!
<br>Arduino app output:<br>
Arduino: 1.8.12 (Linux), Board: "Arduino Uno"<br>
/home/henrik/arduino-1.8.12/arduino-builder returned 139<br>
Error compiling for board Arduino Uno.<br>
<br>
1.8.10 worked fine for what I wanted to do, 1.8.12 didn't. I will try using 1.8.10 arduino-builder with 1.8.12<br>
Can you people at Arduino fix it?

matthijskooijman, 2020-05-07:
Can you try with verbose compilation enabled (there's a checkbox in the preferences, it looks like you have it disabled now) and paste the full output?

facchinm, 2020-05-08:
@HennieP12 can you also provide more info about your configuration (distro version, cpu) ?

HennieP12, 2020-05-09:
@facchinm, I think I fixed it for you...

facchinm, 2020-05-11:
Thanks! The processor is not bleeding edge but it contains all the SSE extensions needed to run modern golang programs (https://github.com/golang/go/wiki/MinimumRequirements#architectures).
@cmaglie could it be a regression introduced with go 1.13 ?