arduino/Arduino #10187: Arduino IDE crash on startup when using Java 14

jcornaz, 2020-05-10:
Using Java 14 the arduino crash with the following exception: 

```
java.lang.ExceptionInInitializerError
        at processing.app.helpers.PreferencesMap.load(PreferencesMap.java:100)
        at processing.app.helpers.PreferencesMap.load(PreferencesMap.java:74)
        at processing.app.PreferencesData.init(PreferencesData.java:56)
        at processing.app.BaseNoGui.initParameters(BaseNoGui.java:834)
        at processing.app.Base.<init>(Base.java:212)
        at processing.app.Base.main(Base.java:150)
Caused by: java.lang.StringIndexOutOfBoundsException: begin 0, end 3, length 2
        at java.base/java.lang.String.checkBoundsBeginEnd(String.java:3756)
        at java.base/java.lang.String.substring(String.java:1902)
        at processing.app.legacy.PApplet.<clinit>(PApplet.java:38)
        ... 6 more
``` 

I am able to launch the arduino IDE only if I tell my system to use Java 11.

facchinm, 2020-05-11:
@jcornaz I think the issue was solved by https://github.com/arduino/Arduino/commit/ca200934721883ff5e86067291fbbc42ca159ef8 , merged in master a few weeks ago. Do you have the chance to test it by self compiling the IDE?

jcornaz, 2020-05-11:
Thanks for your reply @facchinm. No, I don't have a chance to test by compiling the IDE myself.

Is it about to be released soon?

facchinm, 2020-05-11:
Not super soon but in the meantime you can test the hourly build which already contains that commit https://www.arduino.cc/en/Main/Software#hourly

spontarelliam, 2020-05-12:
This change worked for me, thank you!

```
$ java -version
openjdk version "1.8.0_242"
OpenJDK Runtime Environment (build 1.8.0_242-b08)
OpenJDK 64-Bit Server VM (build 25.242-b08, mixed mode)
```

eku, 2020-05-19:
@spontarelliam You write that the changes will work for you but you test with Java 1.8 how you write.

spontarelliam, 2020-05-20:
@eku I won't pretend to understand the endless variety of Java versioning, but I'm running a regularly updated Arch Linux system and was facing this bug for over a year. The change made in ca20093 fixed it.

NicoHood, 2020-07-11:
The fix should be in the new version, right? Can archlinux users please verify that the current version is working with jre14 oder 11? For me it works.

eku, 2020-07-12:
The various Linux distributions have another problem, namely the package dependencies. And that's where it gets really hairy, because there is a distinction between latest and headless and full and so on. Latest rarely points to Java 14.

The dependency of the package of the Arduino IDE to a Java version depends on the packager.

NicoHood, 2020-07-17:
@eku I dont understand what you mean. The issue reported was that arduino crashes with a java version different to java 8 on archlinux.

My first question is, if that issue is solved. The second question is: Could you please describe in detail why this could be a problem for other distributions? I might missunderstand something.

eku, 2020-07-18:
@NicoHood What I actually wanted to say is that the Arduino IDE, as a package of a Linux distribution with nieces, uses the current Java 14, which is why it is difficult to test under Linux. If you install Java 14 and uninstall Java 8, the Arduino IDE will also be uninstalled because of the package dependencies. This is a real crying shame.

NicoHood, 2020-08-02:
On ArchLinux arduino depends on java>=8 so you can use 8,11 or 14. I dont see any problem. On other distributions that is an issue of the distribution/package itself, not arduino and not archlinux. I am asking of the arduino package now works on archlinux. Other maintainers have to adapt as well.

per1234, 2020-08-04:
Hi @jcornaz have you had a chance to try the latest release of the Arduino IDE? Does the issue still occur?

jcornaz, 2020-08-04:
@per1234 It seems to work with latest version of the Arduino IDE.

closed 2020-08-04 by jcornaz