arduino/Arduino #10194: Plase add EmbeddedEducation Library

jsmith2021Brandeis, 2020-05-12:
Please add my organization's Embedded Education Library https://github.com/BrandeisMakerLab/Arduino_Education

It uses curated example files and classes to teach Arduino to Brandeis University students and scientists

per1234, 2020-05-13:
The library was already added last July in https://github.com/arduino/Arduino/issues/9042

There are a couple of things that have caused the tags since 1.0.0 to be rejected by the Arduino Library Manager indexer:
- Neglecting to update the library.properties `version` value before making the release. Each tag must have a unique `version` value.
- Addition of a .development flag file to the library. Any tag which contains this file is rejected.

I see that you have also changed the library.properties `name` value to `EmbeddedEducation` since the last release, which will cause future releases to be rejected. If you want to make the name change, after making your next compliant tag/release you can submit an issue here requesting to have the library name changed.

At the present time, I don't think any action is needed on Arduino's part, so I'll go ahead and close this.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

jsmith2021Brandeis, 2020-05-13:
Thank you, I removed the .development file, and I will update the version before every release

I made this request to add the library because the #9042 issue was a repository on [my personal github](https://github.com/jsmith2021Brandeis/Arduino_Education), and that has moved to an [organizational repo ](https://github.com/BrandeisMakerLab/Arduino_Education/)

Could you either update the repository in the original library or add this new one with the organizational repo and delete the old library?

per1234, 2020-05-13:
OK, I see now. My validation showed the library name was already taken and I saw that you were the one who had made the previous request, but didn't carefully check the repository URLs.

I have reopened the issue and edited your issue so that the new request is up top.

cmaglie, 2020-05-15:
Done.

closed 2020-05-15 by cmaglie