arduino/Arduino #10212: Missing code in MKRRGB.cpp

scottha, 2020-05-15:
The sketches that demonstrate code used with the MKRRGB board fail to compile due to missing symbol definitions used in the SPI_MATRIX call on line 25 of code in MKRRGBMatrix.cpp. There are no definitions of the symbols sercom0, SPI_PAD_)_SCK_1, and SERCODE_RX_PAD_0.

per1234, 2020-05-15:
These are declared in the Arduino SAMD Boards core and the SAMD board variants:
- https://github.com/arduino/ArduinoCore-samd/blob/46f6cad46b55aab1e6081bc1bb60e4fd66dad4c0/variants/mkrzero/variant.cpp#L171
- https://github.com/arduino/ArduinoCore-samd/blob/46f6cad46b55aab1e6081bc1bb60e4fd66dad4c0/cores/arduino/SERCOM.h#L104
- https://github.com/arduino/ArduinoCore-samd/blob/46f6cad46b55aab1e6081bc1bb60e4fd66dad4c0/cores/arduino/SERCOM.h#L75

This library was written specifically for the SAMD architecture of the Arduino MKR, Nano 33 IoT, and Zero boards. If you check the error output, you'll see a helpful warning:
```
WARNING: library Arduino_MKRRGB claims to run on samd architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
```

closed 2020-05-15 by per1234

scottha, 2020-05-15:
Ah. This wasn't at all clear from the web site. I thought the boards could be used by attaching them to an interface shield and then plugging that shield onto an Uno or Mega or whatever. I very much appreciate your efforts to clarify. I couldn't make sense of that error message.

scottha, 2020-05-15:
Thank you very much for your explanation. I wasn’t aware that there was this variant, I very much appreciate your time spent on this, although I have 6 books on Arduino, the documentation still seems to be lacking. I’ve spent 4 days trying to figure out that compile error and what was causing it. Again, thank you.