arduino/Arduino #10214: [IDE] Slow Startup - because of events trigged by 'boardsCustomMenus'

ricardojlrufino, 2020-05-15:
I am doing some analysis to improve the IDE startup time (I am discussing the topic in the email list).

I found another problem that is related to the menus. In particular: `createBoardMenusAndCustomMenus`, and in the method: `filterVisibilityOfSubsequentBoardMenus`.

What happens is that for each menu on a board such as Esp8266, it is calling `LibrariesIndexer.rescanLibraries()`, and it ends up costing a lot of time at startup and also when selecting a board.

This is the stack (not an error ...):

> at processing.app.Base.onBoardOrPortChange (Base.java:1346)
> at processing.app.**Base $ 12.actionPerformed** (Base.java:1605)
> at processing.app.Base.filterVisibilityOfSubsequentBoardMenus (Base.java:1651)
> at processing.app.Base.access $ 0 (Base.java:1635)
> at processing.app.**Base $ 11.actionPerformed** (Base.java:1576)
> at processing.app.Base.rebuildBoardsMenu (Base.java:1550)
> at processing.app.Base. <init> (Base.java:285)
> at processing.app.Base.main (Base.java:150)

As you can see for each menu item, he is manually calling 'actionPerformed', and doing a lot of unnecessary things
```
Set log4j store directory /home/ricardo/.arduino15
SLF4J: Class path contains multiple SLF4J bindings.
SLF4J: Found binding in [jar:file:/media/ricardo/Dados/Workspace/Arduino/arduino-master/arduino-core/lib/slf4j-simple-1.7.22.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: Found binding in [jar:file:/media/ricardo/Dados/Workspace/Arduino/arduino-master/app/lib/slf4j-simple-1.7.22.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: See http://www.slf4j.org/codes.html#multiple_bindings for an explanation.
SLF4J: Actual binding is of type [org.slf4j.impl.SimpleLoggerFactory]
(DEBUG)>>>>>>>>> menu javax.swing.JRadioButtonMenuItem[,0,0,0x0,invalid,alignmentX=0.0,alignmentY=0.0,border=javax.swing.plaf.synth.SynthBorder@48d7ad8b,flags=256,maximumSize=,minimumSize=,preferredSize=,defaultIcon=,disabledIcon=,disabledSelectedIcon=,margin=javax.swing.plaf.InsetsUIResource[top=0,left=0,bottom=0,right=0],paintBorder=false,paintFocus=false,pressedIcon=,rolloverEnabled=false,rolloverIcon=,rolloverSelectedIcon=,selectedIcon=,text=115200]
(DEBUG)>>>>>>>>> onBoardOrPortChange main
java.lang.Throwable
	at processing.app.Base.onBoardOrPortChange(Base.java:1346)
	at processing.app.Base$12.actionPerformed(Base.java:1605)
	at processing.app.Base.filterVisibilityOfSubsequentBoardMenus(Base.java:1651)
	at processing.app.Base.access$0(Base.java:1635)
	at processing.app.Base$11.actionPerformed(Base.java:1576)
	at processing.app.Base.rebuildBoardsMenu(Base.java:1550)
	at processing.app.Base.<init>(Base.java:285)
	at processing.app.Base.main(Base.java:150)
(DEBUG)>>>>>>>>> rescanLibraries main
(DEBUG)>>>>>>>>> rescanLibraries main
(DEBUG)>>>>>>>>> menu javax.swing.JRadioButtonMenuItem[,0,0,0x0,invalid,alignmentX=0.0,alignmentY=0.0,border=javax.swing.plaf.synth.SynthBorder@a53bb6f,flags=256,maximumSize=,minimumSize=,preferredSize=,defaultIcon=,disabledIcon=,disabledSelectedIcon=,margin=javax.swing.plaf.InsetsUIResource[top=0,left=0,bottom=0,right=0],paintBorder=false,paintFocus=false,pressedIcon=,rolloverEnabled=false,rolloverIcon=,rolloverSelectedIcon=,selectedIcon=,text=80 MHz]
(DEBUG)>>>>>>>>> onBoardOrPortChange main
java.lang.Throwable
	at processing.app.Base.onBoardOrPortChange(Base.java:1346)
	at processing.app.Base$12.actionPerformed(Base.java:1605)
	at processing.app.Base.filterVisibilityOfSubsequentBoardMenus(Base.java:1651)
	at processing.app.Base.access$0(Base.java:1635)
	at processing.app.Base$11.actionPerformed(Base.java:1576)
	at processing.app.Base.rebuildBoardsMenu(Base.java:1550)
	at processing.app.Base.<init>(Base.java:285)
	at processing.app.Base.main(Base.java:150)
(DEBUG)>>>>>>>>> rescanLibraries main
(DEBUG)>>>>>>>>> rescanLibraries main
(DEBUG)>>>>>>>>> menu javax.swing.JRadioButtonMenuItem[,0,0,0x0,invalid,alignmentX=0.0,alignmentY=0.0,border=javax.swing.plaf.synth.SynthBorder@6759f091,flags=256,maximumSize=,minimumSize=,preferredSize=,defaultIcon=,disabledIcon=,disabledSelectedIcon=,margin=javax.swing.plaf.InsetsUIResource[top=0,left=0,bottom=0,right=0],paintBorder=false,paintFocus=false,pressedIcon=,rolloverEnabled=false,rolloverIcon=,rolloverSelectedIcon=,selectedIcon=,text=26 MHz]
(DEBUG)>>>>>>>>> onBoardOrPortChange main
java.lang.Throwable
	at processing.app.Base.onBoardOrPortChange(Base.java:1346)
	at processing.app.Base$12.actionPerformed(Base.java:1605)
	at processing.app.Base.filterVisibilityOfSubsequentBoardMenus(Base.java:1651)
	at processing.app.Base.access$0(Base.java:1635)
	at processing.app.Base$11.actionPerformed(Base.java:1576)
	at processing.app.Base.rebuildBoardsMenu(Base.java:1550)
	at processing.app.Base.<init>(Base.java:285)
	at processing.app.Base.main(Base.java:150)
(DEBUG)>>>>>>>>> rescanLibraries main
(DEBUG)>>>>>>>>> rescanLibraries main
(DEBUG)>>>>>>>>> menu javax.swing.JRadioButtonMenuItem[,0,0,0x0,invalid,alignmentX=0.0,alignmentY=0.0,border=javax.swing.plaf.synth.SynthBorder@14a54ef6,flags=256,maximumSize=,minimumSize=,preferredSize=,defaultIcon=,disabledIcon=,disabledSelectedIcon=,margin=javax.swing.plaf.InsetsUIResource[top=0,left=0,bottom=0,right=0],paintBorder=false,paintFocus=false,pressedIcon=,rolloverEnabled=false,rolloverIcon=,rolloverSelectedIcon=,selectedIcon=,text=512K (no SPIFFS)]
(DEBUG)>>>>>>>>> onBoardOrPortChange main
java.lang.Throwable
	at processing.app.Base.onBoardOrPortChange(Base.java:1346)
	at processing.app.Base$12.actionPerformed(Base.java:1605)
	at processing.app.Base.filterVisibilityOfSubsequentBoardMenus(Base.java:1651)
[... 216 lines cut ...]
	at processing.app.Base.main(Base.java:150)
(DEBUG)>>>>>>>>> rescanLibraries main
(DEBUG)>>>>>>>>> rescanLibraries main
(DEBUG)>>>>>>>>> onBoardOrPortChange main
java.lang.Throwable
	at processing.app.Base.onBoardOrPortChange(Base.java:1346)
	at processing.app.Base$12.actionPerformed(Base.java:1605)
	at processing.app.Base.rebuildBoardsMenu(Base.java:1550)
	at processing.app.Base.<init>(Base.java:285)
	at processing.app.Base.main(Base.java:150)
(DEBUG)>>>>>>>>> rescanLibraries main
(DEBUG)>>>>>>>>> rescanLibraries main
(DEBUG)>>>>>>>>> onBoardOrPortChange main
java.lang.Throwable
	at processing.app.Base.onBoardOrPortChange(Base.java:1346)
	at processing.app.Base.<init>(Base.java:294)
	at processing.app.Base.main(Base.java:150)
(DEBUG)>>>>>>>>> rescanLibraries main
(DEBUG)>>>>>>>>> rescanLibraries main
>>>>> pdeKeywords = processing.app.syntax.PdeKeywords@63a28987
>>>>> pdeKeywords = processing.app.syntax.PdeKeywords@63a28987
>>>>> pdeKeywords = processing.app.syntax.PdeKeywords@63a28987
>>>>> pdeKeywords = processing.app.syntax.PdeKeywords@63a28987

```

ricardojlrufino, 2020-05-15:
What I notice in the IDE is that there are a series of problems to keep the menus filled and always consistent, I think it would be better to have a BoardSettingsDialog, to configure the board.

I did something similar with the examples, which had this problem, and were very slow to load.

[image]

ricardojlrufino, 2020-05-16:
See Time lost for loading board Esp8266
On startup the method filterVisibilityOfSubsequentBoardMenus,  (sum the time of all the submenus)
```
 ##### Upload Speed - time: 1612ms  -- thread:pool-1-thread-1
 ##### CPU Frequency - time: 1757ms  -- thread:pool-1-thread-1
 ##### Crystal Frequency - time: 2012ms  -- thread:pool-1-thread-1
 ##### Flash Size - time: 1551ms  -- thread:pool-1-thread-1
 ##### Flash Mode - time: 1571ms  -- thread:pool-1-thread-1
 ##### Flash Frequency - time: 961ms  -- thread:pool-1-thread-1
 ##### Reset Method - time: 1002ms  -- thread:pool-1-thread-1
 ##### Debug port - time: 654ms  -- thread:pool-1-thread-1
 ##### Debug Level - time: 1022ms  -- thread:pool-1-thread-1
 ##### lwIP Variant - time: 850ms  -- thread:pool-1-thread-1
 ##### Builtin Led - time: 895ms  -- thread:pool-1-thread-1
 ##### Erase Flash - time: 1051ms  -- thread:pool-1-thread-1
```

That's because it selects all the submenus, and forces a scan of the libraries.

ricardojlrufino, 2020-05-17:
I managed to correct these errors, so I will do a PR to be able to better evaluate

matthijskooijman, 2020-05-18:
TBH, the board selection code, especially wrt to option menus, is a bit of a mess that could use a big refactor and cleanup. However, I suspect that in the near feature, some of this code might end up being moved into `arduino-cli`, once the java IDE starts to use that (then maybe the resolution of all available board options might be done in `arduino-cli`, with the IDE just showing whatever menus `arduino-cli` returns for the selected board). I'm not sure if this is the plan, maybe @cmaglie can comment.

Anyway, I think fixing this slow startup *would* be good to do already (better to not wait for `arduino-cli` with that, just maybe try it without too much a refactor).

Thanks for looking into this, I'm looking forward to a PR :-)