arduino/Arduino #10215: [Library Manager] Please add https://github.com/cesanta/mjson

cpq, 2020-05-15:
https://github.com/cesanta/mjson is a JSON parser, emitter and JSON-RPC server for embedded systems.

per1234, 2020-05-16:
Hi @cpq. In order to be added to the Library Manager index your library must contain a library.properties file. The format of this file is documented here:
https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format

After adding the library.properties file, you will need to create a new tag/release or redo the existing one. Make sure that the `version` value matches the tag of the release that contains it.

Once you have done this please comment here so we know we can proceed.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

cpq, 2020-05-16:
Thank you for the reminder. Added `library.properties`, please take a look again.

cpq, 2020-05-18:
Is there anything else missing?

cmaglie, 2020-05-18:
Done.

closed 2020-05-18 by cmaglie