arduino/Arduino #10217: [Library Manager] Please add ArduinoBootstrapper

sblantipodi, 2020-05-16:
https://github.com/sblantipodi/arduino_bootstrapper

per1234, 2020-05-16:
Hi @sblantipodi. There are a couple of problems that will prevent your library from being added to the Arduino Library Manager index:

### Use of symlinks

Libraries that use symlinks are prohibited from being included in the Library Manager index:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-is-the-library-list-generated

### Not a valid Arduino library

PlatformIO permits folder structures that are not supported by the official Arduino development software.

I know some people really like PlatformIO, but IMO it's not appropriate to advertise something as "Arduino" if it doesn't even work with the Arduino IDE. Even if you don't want to use the Arduino development software, you should still test to see if the users will be able to do so. You can do this in a lightweight manner with [Arduino CLI](https://github.com/arduino/arduino-cli).

You can get some information about what's required to be a valid Arduino library from the [Arduino Library Specification](https://arduino.github.io/arduino-cli/library-specification/).

---
If you would like your library added to the Library Manager index, please resolve these issues, create a new tag/release, then comment here so we'll know we can proceed. Otherwise, you can close this issue.

sblantipodi, 2020-05-16:
@per1234 
I'm really sorry, I will test the lib from now on every change I made.
last update broke the Arduino compatibility, now it's all fixed in the v1.4.0 release/tag.

I tested it on Arduino IDE and it works like a charm. 
thanks for your support and for your great work here per1234.

thanks

per1234, 2020-05-16:
Thanks! Unfortunately, I have some bad news. I forgot to mention one issue in my previous reply:

Arduino has a policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format

Then update the `version` value in library.properties and create a new release/tag, or else redo the existing release/tag.

Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

Once you have done this, please comment here so we'll know we can proceed.

---
I apologize for the inconvenience I have caused.

sblantipodi, 2020-05-16:
this is bad news since I have just published the same lib in the platformIO registry with the same name.

sblantipodi, 2020-05-16:
what should I do now? @per1234

sblantipodi, 2020-05-16:
@per1234 can I call it simply 
Bootstrapper?

per1234, 2020-05-16:
>can I call it simply Bootstrapper?

Yes. That name is available and valid.

sblantipodi, 2020-05-16:
@per1234 ok thanks,
just released 1.4.3 with that name :)

sblantipodi, 2020-05-16:
just updated to 1.4.4
I'm sorry, last question, when the lib will be available?

thanks for all your help per1234, I appreciate it.

per1234, 2020-05-16:
cmaglie occasionally comes through to add the libraries from the outstanding Library Manager inclusion requests to the index and closes those issues when that is done. After that, it takes some hours for the indexer to pick up the releases for the newly added libraries and for the updated [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache.

>thanks for all your help per1234, I appreciate it.

You're welcome. Thank you for your contribution to the Arduino project!

cmaglie, 2020-05-18:
Done.

closed 2020-05-18 by cmaglie